robotology/icub-models-generator #8: wrong URDF fine calibration configuration

iron76, 2015-03-24:
When putting the urdf-iCub in the ``fine calibration'' configuration (http://wiki.icub.org/wiki/ArmFineCalibration) there is an apparent error in the (left and right) shoulder joint position. This is probably due to a wrong alignment of the torso with the arms.

traversaro, 2015-03-24:
A robust way to proceed is to add to the `icub-tests` suite a consistency check on the transform provided used in iCub. The consistency check can test both `iKin` - `iDyn` consistency, and also `iKin` - `iDynTree` consistency. In this way, we already have a future place for testing consistency regardless of the how this model are produced (for example if in the future we will produce iKin chains from URDF the test will check that the converter is properly working.

traversaro, 2015-03-24:
Unable to reproduce. On my machine, `iKin`/`iDyn`/`URDF` results are consistent,
Furthermore, on Gazebo the arm fine calibration joint positions match the expected result as reported in http://wiki.icub.org/wiki/ArmFineCalibration . (See attached image). 
[image]

closed 2015-03-24 by traversaro

iron76, 2015-03-25:
Covers are however rotated (see figure below) but kinematics seems to be consistent. Is this slight misalignment affecting any other component?

[image]

traversaro, 2015-03-25:
Cover meshes are just used in simulation for display and collision checking. Given that for now we don't have any use case regarding contacts on the `*_upper_arm`, it is a just a visualization issue.