robotology/icub-models-generator #13: Creating URDF for a modified iCub

serena-ivaldi, 2016-07-27:
Hi all,
we would like to create the URDF also for a modified iCub with modified arms. We have the design of the arms in Solidworks. We would like to attach them to an iCub model to make a full new robot.
Could you help us in opening the full robot in Solidworks? The original files do not seem compliant.

Thanks in advance

@bpapaspyros @jb_mouret @avoilque @dgoepp

bpapaspyros, 2016-08-05:
Hi,

Any news on that ? To be more specific, we want to replace the arms and produce a new URDF file with the new parts. Patching the existing URDF or even designing the new parts may not be as accurate as we want without the original Solidworks designs, hence we opened this issue. 

Thank you,
@serena-ivaldi @jbmouret @Anthonyfma @dogoepp

traversaro, 2016-08-05:
Sorry, I was a bit stressed this two weeks due to some deadlines. : ) 
Can we discuss your requirements more in detail next week?

traversaro, 2016-08-05:
One thing that I can already discuss is that there are no "original Solidworks designs", as all the iCub CAD design is done in PTC CREO Parametric, see http://wiki.icub.org/wiki/ICub_mechanics_CAD .

bpapaspyros, 2016-08-05:
> Sorry, I was a bit stressed this two weeks due to some deadlines. : ) 
> Can we discuss your requirements more in detail next week

No worries. Sure we can discuss it in depth next week ! 

> One thing that I can already discuss is that there are no "original Solidworks designs", as all the iCub CAD design is done in PTC CREO Parametric, see http://wiki.icub.org/wiki/ICub_mechanics_CAD .

Pardon me, I meant to say we want to integrate the original designs with Solidworks (if there is a way to do that or even take a different approach altogether if you can guide us), not that they are Solidworks designs necessarily. Thank you for the link, we will take thorough look and update you on the details.

Thank you

@serena-ivaldi @jbmouret @Anthonyfma @dogoepp

traversaro, 2016-08-09:
A URDF model generated specifically for the `iCubNancy01` has been generated in https://github.com/robotology-playground/icub-model-generator/issues/14 . Apparently you have the forearm v1 (just in case, can you send some photos to make sure that we are getting the right version?) so we need to create the model using the `dh` model creation pipeline, as we don't have a simmechanics model for a robot with v1 arms.

traversaro, 2016-08-09:
@bpapaspyros @dogoepp Can we discuss on Skype (or where you prefer) your specific requirements? My Skype id is `silvio.traversaro`.

traversaro, 2016-08-22:
Sorry for the late update @bpapaspyros , I was able to talk with some mechanical guys and they told me that generating a "reduced" version of iCub that could be imported in SolidWorks is not trivial, even in "simple" formats such as STEP or STL. 
In the meanwhile, I thought an alternative solution: if we could precisely define a frame in the iCub part on which you mount the modified forearm, you can define the corresponding frame in your SolidWorks assembly and then we would have all the necessary information to modify the URDF. 

Said that, could you provide some more info on the part on which you mount the modified forearm?

This is an explosion of the forarm, I don't know if you attach the modified forearm on one of these parts or to some parts: 
[image]

traversaro, 2016-08-22:
If you need to explore the CAD of iCub to identify the part that you use to mount the modified forearm, you can use the free Creo View Express : http://www.ptc.com/cad/creo/view-mcad/extension-express .

Anthonyfma, 2016-08-22:
Hi,

I'm Anthony, the student who design the forearm prototype.
The new forearm replace all parts of the iCub forearm. It's attached on the elbow part (instead of the left cyan part on the explosion CAD).
[image]
[image]

I hope it will be a bit more clear.

traversaro, 2016-09-09:
Hi everyone, sorry for the late reply.

I discussed with the mechanical guys, and we agreed that the easier way to combine the model it that we agree on a a frame define on the interface between the "regular" iCub and your part, and using that frame we mount through software your part on the robot URDF.

We identified the following frame:
[image]
[image]
[image]

We can call this `l_custom_forearm_mounting_base` and `r_custom_forearm_mounting_base` (attached to the `l_elbow_1`/`r_elbow_1` link, see http://wiki.icub.org/wiki/ICub_Model_naming_conventions ). 

I would ask you to add this frame on your custom forearm (i.e. where this frame is located on your custom part when the forearm is mounted on the robot) and provide the following information expressed in this frame: 
* Inertial information in URDF format ( see http://wiki.ros.org/urdf/XML/inertial, for more info check out http://gazebosim.org/tutorials?tut=inertia&cat= ) that can be easily obtained by SolidWorks: 
  * The total mass of the part. This can be obtained by SolidWorks, but I suggest that you double check by weighing the piece with a weighing machine. 
  * The center of mass position with respect to the `*_custom_forearm_mounting_base` origin and with the orientation of the `*_custom_forearm_mounting_base`.
  * The 3x3 Inertia Matrix of the object, with respect to the center of mass and with the orientation of `*_custom_forearm_mounting_base`. 
* STL Mesh of the part : you are the expert here, but I suggest that you provided a simplified one for collision and a more complex one (perhaps even in Collada `dae` format if you want to have nice colors) for visualization.

Anthonyfma, 2016-09-09:
Hi,

It won't be so easy to obtain reliable informations about the new forearm. As it is 3d printed, it is not homogeneous. So an important thing is that Solidworks data are going to be approximative. Moreover, the once the skin is added, properties will change again. We will manage to approximate as precisely as possible.

Best regards,

traversaro, 2016-09-09:
Yes, for this reason I was suggesting to weight the new forearm (with also the skin and everything that is going to be added on top). Once you have the mass, assuming the density homogeneous is typically (in my experience) a good enough approximation. 
So you can: 
  * Compute COM and Inertia from the model or from the meshes using `MeshLab`, as documented in http://gazebosim.org/tutorials?tut=inertia&cat= . 
  * Scale COM and Inertia to match the new mass (this is straighforward from the COM and inertia equations.

traversaro, 2016-11-08:
@serena-ivaldi Just a remainder to avoid misunderstanding : on this activity IIT is waiting for feedback from Nancy.

dogoepp, 2017-01-13:
Hi and sory for the late reply,

I'm working on getting the required measurements and models. I would just need one more information in order to properly define the frame `*_custom_forearm_mounting_base`: the distance shown in red lines in the image bellow.

[image]

Thanks a lot

traversaro, 2017-01-13:
Hi @fiorisi , when you have time could you help me getting the data requested by @dogoepp ? 
Thanks

Anthonyfma, 2017-01-13:
Hello Dorian.
On the folder we discussed about, there is a modelisation of iCub upper
body. You will find the part, and measure it directly on Solidworks.

Regards,

fiorisi, 2017-01-23:
Hi @dogoepp, I have edited the previous comment of @traversaro because when I defined the frame I have used the wrong iCub model. Please update your transformation matrix. Sorry for the mistake, let me know if you need further help.

dogoepp, 2017-01-24:
Thanks @fiorisi, this should be enought for me. I hope to be able to work on this next Friday.

dogoepp, 2017-02-03:
Based on the changes made by @fiorisi, I got the following mass properties from Solidworks

Mass property of the fore-arm

Mass = 199.77 gramms

Volume = 199765.57 cubic millimeters

Surface = 78439.82  square millimeters

Center of mass: ( millimeters )
	X = 0.01
	Y = -2.06
	Z = 90.71

Pricipal axes and moment of inertia: ( gramms *  square millimeters )
At center of mass
	 Ix = ( 0.00, -0.02,  1.00)   	Px = 72761.58
	 Iy = ( 0.01, -1.00, -0.02)   	Py = 584460.80
	 Iz = ( 1.00,  0.01,  0.00)   	Pz = 587155.38

Moments of inertia: ( gramms *  square millimeters )
At center of mass and aligned with the reference frame.
	Lxx = 587154.95	Lxy = -26.28	Lxz = 394.66
	Lyx = -26.28	Lyy = 584242.38	Lyz = -10572.54
	Lzx = 394.66	Lzy = -10572.54	Lzz = 72980.42

Moments of inertia: ( gramms *  square millimeters )
In the reference frame.
	Ixx = 2231904.66	Ixy = -29.13	Ixz = 519.87
	Iyx = -29.13	Iyy = 2228141.10	Iyz = -47975.01
	Izx = 519.87	Izy = -47975.01	Izz = 73831.42

I assume that what you're most interested in is the latest matrix. I still have to weight the actual fore-arm with the skin attached and then correct the values of these matrices.

dogoepp, 2017-02-24:
We are starting to modify the mechanical design of the fore-arm in order to make it easier to mount and to make it possible to plug the skin sensors. Therefore, I'll probably need to update the matrices and the weights I had measured will no longer hold.

However, for the record, the full forearm with the skin weights 151g.

dogoepp, 2017-02-24:
Actually, the changes we would like to make to the plastic fore-arm require to have the coordinates and orientations of the holes to attach the skin (and the plastic support) to the arm.

Do you have an advice on how we could get this information ?

traversaro, 2017-05-19:
Sorry for the late reply @dogoepp . I will answer your question later today.

traversaro, 2017-06-15:
Sorry for taking another month to reply. If I understand correctly, you need the location of the holes to which the screws of the skin are mounted on the internal part of the forearm. An STL of the forearm without the covers would be sufficient for you? 

cc @fiorisi

dogoepp, 2017-06-16:
Hi @traversaro. You understand correctly. We managed to get approximate coordinates from the STL of the fore-arm without cover but I forgot to update the mass properties.

Once I have the new mass properties I think that we are good for this part and can move forward to make iCub work with this fore-arm and build the related URDF.