robotology/icub-models-generator #20: Simplifying the sensor frame "pose" information in the URDF

nunoguedelha, 2016-08-14:
@traversaro , as soon as the accelerometers sensor data will be integrated in the URDF, we will get 3 equal instances of each sensor frame pose:
1. in the <sensor> tag
2. in the <gazebo> tag
3. in the fake link frame attached to the reference frame through a fixed joint.

If I understood well, the fake link frame is useful for iDynTree, such that we don't need another pose information under the respective <sensor> tag. We could have, for instance, for a given sensor:
```
  <link name="l_forearm_mtb_acc_1b9"/>
  <joint name="l_forearm_mtb_acc_1b9_fixed_joint" type="fixed">
    <origin xyz="0.0595957174403 0.0596569 -0.0205995080872" rpy="1.76682134218 0.261099829847 3.08752548202"/>
    <parent link="l_elbow_1"/>
    <child link="l_forearm_mtb_acc_1b9"/>
    <dynamics damping="0.1"/>
  </joint>
  ...
  <gazebo reference="l_forearm_mtb_acc_1b9">
    <sensor name="l_forearm_mtb_acc_1b9" type="imu">
      <always_on>1</always_on>
      <update_rate>100</update_rate>
      <pose>0.0 0.0 0.0 0.0 0.0 0.0</pose>
    </sensor>
  </gazebo>
  ...
  <sensor name="l_forearm_mtb_acc_1b9" type="accelerometer">
        <pose>0.0 0.0 0.0 0.0 0.0 0.0</pose>
	<parent link="l_forearm_mtb_acc_1b9"/>
  </sensor>

```

or even remove the `<pose>` tag in `<sensor>` and `<gazebo>`. It would make a lighter URDF with less repetition and less confusion. What do you think?

traversaro, 2016-08-14:
I got the point about reducing the information duplication. 

Consider anyhow that since https://github.com/robotology/idyntree/commit/ca27f6906c4787df49bee134eb7563d3a8a42245 , the sensor frames are automatic added as additional frames when creating a `iDynTree::Model` from an URDF file (see https://github.com/robotology/idyntree/blob/master/src/model_io/urdf/include/iDynTree/ModelIO/URDFModelImport.h#L31) so we could get rid of one duplication of the fake link with the name of the sensor. 

Furthermore,  currently URDF sensor parsed by iDynTree need to be attached to "real" links, not to "fake" links, as the `link` attribute of the `parent` tag is searched among the `iDynTree::Model` links (see https://github.com/robotology/idyntree/blob/master/src/model_io/urdf/src/URDFGenericSensorsImport.cpp#L252). We can modify this to check for the frame and get the link to which the frame is attached, but I think it is convenient to explicitly attach the sensor in the URDF to the actual link to which it is attached, otherwise it could be confusing for users, as the sensor `l_forearm_mtb_acc_1b9` would be attached to (fake) link `l_forearm_mtb_acc_1b9` in the URDF, but attached to link `l_forearm` in the `iDynTree::Model`.

nunoguedelha, 2017-03-12:
@traversaro I will then PR the YAML config file updates with the sensors export frame option deactivated (flags `exportFrameInURDF` set to No).

nunoguedelha, 2017-06-21:
Solved by https://github.com/robotology-playground/icub-model-generator/issues/49.

closed 2017-06-21 by nunoguedelha