robotology/icub-models-generator #25: Generate gazebo models

traversaro, 2016-08-16:
We need to generate a Gazebo model for each major revision of the iCub, see https://github.com/robotology/icub-gazebo/issues/16 .

traversaro, 2017-06-22:
In the current generation process in place, it should be trivial to generate a Gazebo model for any specific iCub instance (then exposing just some "major" models for simplifying the life to our users is doable as well). 

We discuss a bit on how to implement this with @nunoguedelha today, updating the issue to avoid to forget. 
We would like to use the structure that we set up in https://github.com/robotology-playground/icub-models . Currently in that repo the URDF can be used by both ROS and YARP software with no problem. Using them directly in gazebo is a bit problematic, so the options are: 
* Automatically generate a separate Gazebo model directory, copying at build/install time the meshes, and add that one to `GAZEBO_MODEL_PATH`,
* hack a bit the system, and keep the existing structure without copying meshes in a different position, so that models can also be used directly from the source directory.

traversaro, 2017-06-22:
Relatively to the second option, let's take note of somethings. 
A Gazebo model is identified by a first-level directory containing a `model.config` inside one of the directories specified in the `GAZEBO_MODEL_PATH` environmental variables [1]. 
This means that the `iCub` directory in https://github.com/robotology-playground/icub-models can't be put directly in `GAZEBO_MODEL_PATH`, because the models are inside the `robots/iCub***` directory (and this is a constraint of the YARP system to handle robot-specific configuration files). 
A possible solution to this is to add `iCub/robots` to the GAZEBO_MODEL_PATH. 
Once a model is correctly found by Gazebo, the meshes and other assets (such as `.ini` YARP configuration files) need to be found by Gazebo. Gazebo uses a URI scheme, and automatically convert any `package://` scheme to `model://`. The nice thing is that the `model://` identify a file with its location in **any** of the directories contained in `GAZEBO_MODEL_PATH`, not only the directory for which the original model was loaded! 
So, to load assets that are contained in the  `iCub/meshes` directory (and eventually `iCub/gazebo` etc, etc) it is sufficient to add the appropriate directory to `GAZEBO_MODEL_PATH` env variable. 

[1] : https://bitbucket.org/osrf/gazebo/src/b09f8d3594774551e76bd53789eb72b575fb2775/gazebo/gui/InsertModelWidget.cc?at=default&fileviewer=file-view-default#InsertModelWidget.cc-331

traversaro, 2017-06-22:
Related: https://bitbucket.org/osrf/gazebo/pull-requests/2479 .
By the way, I think that the ideal way to solve this kind of problems would be if `URDF`/`SDF` had a relative path URI scheme, to specify the location of assets relatively to the `URDF`/`SDF` file.

nunoguedelha, 2018-01-11:
This issue originated an Epic (https://github.com/robotology-playground/icub-model-generator/issues/52) that is now tracking all the deriving issues and PRs for implementing the model generation process described here.

Closing this issue.

closed 2018-01-11 by nunoguedelha