robotology/icub-models-generator #27: iCub v2.5 without backpack generated URDF models are not valid

traversaro, 2016-08-18:
Stricter checks in iDynTree routines for loading URDF (see https://github.com/robotology/idyntree/pull/212) highlighted  that there are XML validity problems in some generated URDF (see https://travis-ci.org/robotology-playground/icub-model-generator/builds/153229462).

All the models failing are iCub v2.5 without backpack, so the problem is probably in https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/adjust_com_for_model_without_backpack.py . 

~~~
$ ctest --output-on-failure
Test project /home/travis/build/robotology-playground/icub-model-generator/build
    Start 1: iCubGenova03ConsistencyCheck
1/8 Test #1: iCubGenova03ConsistencyCheck ......   Passed    0.02 sec
    Start 2: iCubNancy01ConsistencyCheck
2/8 Test #2: iCubNancy01ConsistencyCheck .......   Passed    0.02 sec
    Start 3: iCubParis01ConsistencyCheck
3/8 Test #3: iCubParis01ConsistencyCheck .......   Passed    0.02 sec
    Start 4: iCubParis02ConsistencyCheck
4/8 Test #4: iCubParis02ConsistencyCheck .......   Passed    0.02 sec
    Start 5: iCubGenova01ConsistencyCheck
5/8 Test #5: iCubGenova01ConsistencyCheck ......***Failed    0.01 sec
[ERROR]  :: modelFromURDFString : Invalid XML
[ERROR] error in loading robot model
icub-model-test error: impossible to load model from 0
    Start 6: iCubGenova02ConsistencyCheck
6/8 Test #6: iCubGenova02ConsistencyCheck ......   Passed    0.02 sec
    Start 7: iCubGenova05ConsistencyCheck
7/8 Test #7: iCubGenova05ConsistencyCheck ......***Failed    0.01 sec
[ERROR]  :: modelFromURDFString : Invalid XML
[ERROR] error in loading robot model
icub-model-test error: impossible to load model from 0
    Start 8: iCubDarmstadt01ConsistencyCheck
8/8 Test #8: iCubDarmstadt01ConsistencyCheck ...***Failed    0.01 sec
[ERROR]  :: modelFromURDFString : Invalid XML
[ERROR] error in loading robot model
icub-model-test error: impossible to load model from 0

63% tests passed, 3 tests failed out of 8
Total Test time (real) =   0.12 sec

The following tests FAILED:

	  5 - iCubGenova01ConsistencyCheck (Failed)
	  7 - iCubGenova05ConsistencyCheck (Failed)
	  8 - iCubDarmstadt01ConsistencyCheck (Failed)

Errors while running CTest

traversaro, 2016-08-18:
It would be convenient to have a xsd to check consistency of generated urdf https://github.com/robotology/idyntree/issues/214 .

francesco-romano, 2016-08-19:
I agree.
I think also this should be checked when the urdf is loaded.
Unfortunately TinyXml does not support this AFAIK

traversaro, 2016-09-10:
The problem was just a missing newline in the last line of the file, fixed in https://github.com/robotology-playground/icub-model-generator/commit/b6f24b03b2e3a510daf52c5ef1c289f19e89dbdf .

closed 2016-09-10 by traversaro