robotology/icub-models-generator #33: Missing raw STL for Simmechanics generated models

traversaro, 2016-10-28:
@bmagyar correctly pointed out that currently the SimMechanics generated models refer to meshes of the type `model://iCubGenova02/meshes/dae/sim_sea_2-5_root_link_prt.dae`. 

This file names are generated by this value : https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml#L4 of the [`filenameformatchangeext`](https://github.com/robotology/simmechanics-to-urdf#mesh-parameters) parameter of `simmechanics-to-urdf`. The actual meshes they are referring to (the original name can be obtained by removing the `filenameformatchangeext` option, or checking the content of the SimMechanics file) are the `.STL` generated by the SimMechanics Link Creo exporter. Those files are enormous, and we are not committing them directly in `icub-model-generator/simmechanics/data/icub2_5/` to avoid to bloat the repo. 
The ideal situation would be to have a semiautonomous pipeline to convert this meshes in something useful, but for the moment providing them in someway in this repo is the basic state to make the generate models actually useful.

traversaro, 2016-10-28:
Related to https://github.com/robotology-playground/icub-model-generator/issues/28 .

traversaro, 2016-10-28:
I searched for the latest generated raw STL but I failed : ( . 
I hope @fiorisi knows where they are stored.

bmagyar, 2016-11-01:
So renaming the references in the urdf according to the yaml file you referenced should make the urdf-s work with the files from [here](https://github.com/robotology-playground/icub-model-generator/tree/master/simmechanics/data/icub2_5/meshes/dae) ?

traversaro, 2016-11-08:
@bmagyar sorry for the delay in replying. 

Yes, those meshes where an outcome of another manual process of mesh generation by @fiorisi  , however there were problems related to colors, but they should work fine beside that. 

One problem is that since then we added the three wrist dofs to the generated model, those meshes do not include those DOFs (i.e. the forearm mesh include also the hand mesh).

traversaro, 2017-02-15:
Until we find a proper way of hosting those meshes, they can be found in this archive: https://www.dropbox.com/s/e2wqa1740r8jacj/icub2d5withBBgeneratedNov2016.zip?dl=0 . The meshes names and frames should be compatible with the latest version of the `ICUB_2-5_BB_SIM_MODEL.xml` model, available at https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_SIM_MODEL.xml .

traversaro, 2017-03-30:
This needs to be solved for the IHMC integration.

traversaro, 2017-04-18:
See https://github.com/gerkey/ros1_external_use#installing-for-use-by-tools-like-roslaunch for how to install the files in a way that are found by the `package://` URL in URDF.

traversaro, 2017-06-29:
Fixed by https://github.com/robotology-playground/icub-model-generator/pull/46 (even if support for `package://` URI is still missing in iDynTree: https://github.com/robotology/idyntree/issues/291 ).

closed 2017-06-29 by traversaro