robotology/icub-models-generator #34: Review torque (effort) limits in generated models

traversaro, 2016-11-10:
@randaz81 noticed that some issues in iCub Gazebo models (see https://github.com/robotology/icub-gazebo/issues/35) where due to misplaced torque effort limits.

traversaro, 2016-11-10:
Related discussion  in Gazebo bugtracker : https://bitbucket.org/osrf/gazebo/issues/780/provide-torque-speed-curve .

traversaro, 2016-11-10:
For simulation purpose, I think it make sense to use the nominal maximum torque of the motor/transmission system. We should document this values in the `csv` file used in generating the model : https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_joint_parameters.csv .

S-Dafarra, 2017-09-22:
I don't know whether this is the right place to point thia out, but I noticed that the effort limits on the knee are too conservative. By putting the robot in a statically balanced position in gazebo, with the knee consistently bent (as it happens during single support phase in static walking), I noticed that the robot was falling due to the torque saturation. By using 50 instead of 30 as effort parameter, everything was working just fine.

traversaro, 2017-09-22:
cc @fiorisi

fiorisi, 2018-05-04:
In [this issue](https://github.com/robotology/QA/issues/256) are collected the velocity and effort limits for the SEAs legs. The limits have been copied in the [models](https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_joint_parameters.csv.in) generated using the [SimMechanics pipeline](https://wiki.icub.org/wiki/Creo_Mechanism_to_URDF). Probably we can close the issue for the time being.