robotology/icub-models-generator #36: Fix materials exported in URDF file

traversaro, 2016-12-13:
Currently all the models exported using the creo --> simmechanics --> urdf workflow do not properly export materials. 
For example if you load of today any URDF in iDynTree, you get  the following warnings:
~~~
`[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
[ERROR]  :: parseURDFMaterial : Impossible to parse URDF material, setting material to white.
~~~

traversaro, 2016-12-13:
When we fix this issue, we can revert this commit: https://github.com/robotology/idyntree/commit/5e760abb90642899d81db803b3e653343c74bb4f .

traversaro, 2017-07-05:
We checked this problem with @diegoferigo and apparently this line of `simmechanics-to-urdf` seems related: https://github.com/robotology/simmechanics-to-urdf/blob/8c23cbcda83687dcb32d83f98a9fbb6b5f221349/simmechanics_to_urdf/firstgen.py#L989 . 
It could be interesting to check upstream (on https://github.com/ros/urdfdom) if you are allowed to refer by name a material that is only defined in a link, rather than in the robot tag.

traversaro, 2017-07-05:
I am  a bit confused. By a quick check it seems that indeed this kind of "global scope for link material names" is not supported by the `urdfdom` reference implementation : 
https://github.com/ros/urdfdom/blob/8b27e0b6fbddbb4cc670156961bd52b4d5ea632b/urdf_parser/src/model.cpp#L146 . 

However, the iCub models seem to be correctly parsed by `urdf_to_graphviz`, so I am a bit confused.

nunoguedelha, 2017-11-09:
ciao @traversaro , do you agree with moving this issue to the backlog for now?
cc @diegoferigo

traversaro, 2017-11-09:
Yes, I think it already easy is in the backlog de facto.

nunoguedelha, 2017-11-10:
ok, updated the pipeline.

traversaro, 2018-02-09:
To be honest, I would simply suppress the warning in iDynTree until we found a way to fix the problem at the generator level.

diegoferigo, 2018-02-11:
This is somehow related to https://github.com/robotology-playground/icub-model-generator/issues/62 if I remember correctly. If colors are properly exported from the  pipeline, the generator would not fill the model with these random colors.

traversaro, 2018-02-12:
Another way to fix this is to make sure that the simmechanics-to-urdf converter does not emits malformed URDF files (referencing to non-existing colors) if the MaterialProp element is not present.

closed 2018-06-20 by traversaro