robotology/icub-models-generator #42: *_upper_leg_back_contact frames

traversaro, 2017-02-17:
@gabrielenava defined and added to the `simmechanics` generated models the `(r/l)_upper_leg_contact` frames, that are convenient for experiments in which the robot is sitting on its legs. After a day, I suggest to renamed this frames to `*_upper_leg_back_contact` to avoid confusion. 

This frames work fine, but they have been just committed to the robot models contained in `yarp-wholebodyinterface` , and I will list here the thing that need to be addressed to have them properly integrated in the model generation pipeline. 

* [ ] Document their location in http://wiki.icub.org/wiki/ICub_Model_naming_conventions#URDF_frames,
* [ ] Add them to Creo model committed in the hardware repository and regenerate the shrinkwraps for the upper_leg links and then regenerate the simmechanics xml,
* [ ] Commit the generate file in `icub-model-generator` and regenerated all the URDFs ,
* [ ] Copy the generated models in `yarp-wholebodyinterface`.

cc @DanielePucci

gabrielenava, 2017-08-28:
I asked for Mechanical engineers support, and they are aware of this issue and our requirements.

gabrielenava, 2017-09-07:
According to https://github.com/robotology-playground/icub-models/blob/master/iCub/robots/iCubGenova04/model.urdf, the new iCub model contains also the leg contact frames.

fiorisi, 2018-05-04:
Updated the documentation in the wiki. As discussed with @gabrielenava, probably `yarp-wholebodyinterface` is deprecated, so we can close the issue? Or do we have to copy the models in another location?

gabrielenava, 2018-06-04:
related issue: https://github.com/robotology/robotology-superbuild/issues/14

traversaro, 2018-08-23:
I noticed that this frames are not present in the iCubNancy01 model, differently from the version manually patched in https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubNancy01/model.urdf . However, I do not know if anyone is using those frames, as the controllers in whole-body-controllers does not seem to support anymore iCubNancy01, at least judging from the configuration file.

gabrielenava, 2018-08-23:
Yes, at the moment the torque controllers only support `icubGazeboSim`, `iCubGazeboV2_5`, and `iCubGenova02/04`. To keep and ordered and maintainable structure of the repo, the idea was to port from `WBIToolbox-Controllers` only the bare essentials, and then to add other robots/features only if necessary. I think for the moment we can procedd without `iCubNancy01` and if necessary we can add it later. Or should we ask to the Nancy lab?

traversaro, 2018-08-23:
cc @DanielePucci

DanielePucci, 2018-08-23:
If it takes too long time to generate a no-battery-pack model with these new frames, we may take the `iCubGenova02/04` are remove some torso mass to generate the `iCubNancy01` model, right?

traversaro, 2018-08-23:
The Creo-generated model without backpack (such as the iCubDarmstadt01) already contains this models (but at the moment the iCubNancy01 robot is still generated with the `dh` parameters hardcoded in the code because it uses arv v1). However, the question was if we want to have iCubNancy01 configuration files for  the controllers. If we don't, I think we can deal with this model problem when we will work in adding back the configuration files to the controllers.

traversaro, 2018-09-12:
I prefer to  solve the iCubNancy01 problem before closing this issue, see also https://github.com/robotology/yarp-wholebodyinterface/issues/89 .