robotology/icub-models-generator #43: Add iCubGenova04

traversaro, 2017-04-06:
It is a classical iCub 2.5 with backpack, so it should be simple.

traversaro, 2017-05-03:
@gabrielenava can you take care of this? It should be just necessary adding some lines in https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/CMakeLists.txt#L44 and probably adding the tests in https://github.com/robotology-playground/icub-model-generator/blob/e3b35c515acd31a1b31119a0dd3f4dcad8a0a82a/tests/CMakeLists.txt#L29 .

francesco-romano, 2017-05-03:
@traversaro without backpack, isn't it?

traversaro, 2017-05-03:
[image] ?

francesco-romano, 2017-05-03:
You know that I was sure that iCubGenova04 was without backpack?!?!?!

traversaro, 2017-06-15:
Done in https://github.com/robotology-playground/icub-model-generator/issues/45 .

closed 2017-06-15 by traversaro