robotology/icub-models-generator #44: urdf icubNancy need meshes

InriaBrice, 2017-04-24:
Hi, 

We got some issues with icubNancy URDF, we need to get some meshes .dae for part of the iCub which are not end effector (l_hip_yaw, r_hip_yaw, torso_yaw, l_shoulder_yaw, r_shoulder_yaw, l_ankle_2, r_ankle_2, l_elbow_1, l_elbow_1).

I need it to compute some convex hull to do automatic collision.
For now i took l_hip_pitch.dae to compute convex hull of l_hip_yaw (same for right leg etc, but i don't know what to take for torso_yaw).

Thanks

traversaro, 2017-05-08:
Hi @InriaBrice , sorry for the delay in the reply. 
I am cleaning up today the meshes generation (issue https://github.com/robotology-playground/icub-model-generator/issues/28) and I will keep you updated.

InriaBrice, 2017-05-24:
Hi @traversaro , thanks, also can we get the foot meshes to walk ? Our robot foot are 20*9 and in our simulation meshes are 15*5.

Best regards,

traversaro, 2017-06-15:
Hi @InriaBrice , sorry for the late reply. 

A faithful meshes of your foot are the following ones: 
* https://github.com/robotology-playground/icub-model-generator/blob/7b4ebcba1ae06d9e8e60de6c80b217f97e66cac3/simmechanics/data/icub2_5/meshes/simmechanics/sim_sea_2-5_l_sole_prt-binary.stl
* https://github.com/robotology-playground/icub-model-generator/blob/7b4ebcba1ae06d9e8e60de6c80b217f97e66cac3/simmechanics/data/icub2_5/meshes/simmechanics/sim_sea_2-5_r_sole_prt-binary.stl

Unfortunately, we are not currently using those meshes for generating the model of iCubNancy01 because iCubNancy01 is an hybrid with most of the body v2.5, but the arms are v1 arms (see https://github.com/robotology/yarp-wholebodyinterface/pull/78#issuecomment-302666016).

InriaBrice, 2017-06-16:
Thanks @traversaro ! The feet was a bit huge lol. But i resized them so it's ok.

traversaro, 2017-06-16:
I think the STL is expressed in millimeters : ) , that is why in the URDF we have a `scale="0.001 0.001 0.001"` attribute to the mesh element.