robotology/icub-models-generator #48: URDF models have non-consistent joint limits

S-Dafarra, 2017-06-16:
This is somehow related to https://github.com/robotology-playground/icub-model-generator/issues/5 .
The limits displayed in [this]( https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_joint_parameters.csv ) file are different from those defined in https://github.com/robotology/robots-configuration/tree/devel . 
For example the ``l_hip_roll`` has different limits from those defined in [this](https://github.com/robotology/robots-configuration/blob/devel/iCubGenova04/hardware/motorControl/left_leg-eb6-j0_3-mc.xml) file. 

We should even discuss whether to use the "soft" or the "hardware" limits. (https://github.com/robotology/robots-configuration/blob/devel/iCubGenova04/hardware/motorControl/left_leg-eb6-j0_3-mc.xml vs https://github.com/robotology/robots-configuration/blob/devel/iCubGenova04/hardware/mechanicals/left_leg-eb6-j0_3-mec.xml)

traversaro, 2017-06-29:
Please note that the limits in the robots-configuration files are sometimes even inconsistent among different robots, even if the different robots are the same version of the iCub (i.e. we should check the limits before copying them). 

Having said that, I think we should have actual **hardware** limits in the URDF. In the `gazebo-yarp-plugins` we have a way to specify additional software limits, so I think that given that the URDF tipically described the hardware of the robot, the limits in the URDF should be the hardware ones. 
Note that this means that user code, such as IK, **should not** use the limits from the URDF, but rather the software limits, for example the one that in YARP you can obtain using the [`IControlLimits`](http://www.yarp.it/classyarp_1_1dev_1_1IControlLimits.html) interface.

traversaro, 2017-06-29:
Once we have the "actual" limits for the iCub 2.5, we can update the file https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_joint_parameters.csv to propagate this changes to all generated models.

traversaro, 2017-06-29:
cc @DanielePucci

DanielePucci, 2017-06-29:
@fiorisi

fiorisi, 2017-07-05:
what about finalizing this [table](http://wiki.icub.org/wiki/ICub_Model_naming_conventions#Joint_additional_information)? can we reorganize the wiki for the different iCub models?

traversaro, 2017-07-05:
👍 for finalizing that table in the current location, differentiating it for iCub version if necessary.

fiorisi, 2018-05-04:
Recently the [iCub Manual](http://wiki.icub.org/wiki/Manual) has been updated with a [page about the joint limits](http://wiki.icub.org/wiki/ICub_joints_limits). We could use the `CAD limits` as the limits for the model, the problem is that we currently have these limits only for the legs. Alternatively, we could use the `Hardware limits`, but the RoM for some joints (e.g. ankle) is not the widest. What do you suggest?

traversaro, 2018-08-23:
I would go for CAD RoM, while the software limits should go in the Gazebo configuration files. However it is commont practice to use the URDF limits for planning movements, so we should make sure no one is actually doing that in the lab, and instead is reading the joint limits from the robot at runtime.