robotology/icub-models-generator #52: Use the generated model in Gazebo

diegoferigo, 2017-07-14:
In this issue I want to track and recap the efforts done (mainly by @nunoguedelha) and what's missing to let these model work in Gazebo.

- [ ] Creating a first version that simulates (https://github.com/robotology-playground/icub-model-generator/pull/49). These changes for now have been applied **manually** to the `icub-models` repository (https://github.com/robotology-playground/icub-models/pull/2)
- [ ] Install the `model.config` for all the robots
- [ ] Enable all the `inertials`. Right now gazebo crashes without giving any error
- [ ] Handle the name of the robot (https://github.com/robotology/simmechanics-to-urdf/issues/24)

cc @nunoguedelha @traversaro @DanielePucci

traversaro, 2017-10-24:
As discussed f2f with @DanielePucci , @francesco-romano and @fiorisi , for unblocking this issue the plan is to automatically change the inertia of the small links using the `assignedInertia` feature of simmechanics-to-urdf script, and by enabling the implicit damping (see https://github.com/robotology-playground/icub-model-generator/issues/24).

traversaro, 2017-10-24:
Yes, but I think that the enabling the implicit spring damping should decrease the amount to which the inertia should be increased.

nunoguedelha, 2017-10-24:
Regarding the point "Handling the small inertias." in the checklist, we changed the inertia of some of the smallest robot links. This is to be ported into the URDF generation scripts. The impacted links are listed below, with the respective inertia delta. All the changed link inertia have been set to `<inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>` as a quick fix:
```
<link name="r_hip_3">
-      <inertia ixx="3.27938e-05" ixy="2.67901e-07" ixz="-4.41463e-09" iyy="5.69957e-05" iyz="3.32338e-08" izz="3.26449e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_ankle_2">
-      <inertia ixx="0.000335682" ixy="-7.53335e-08" ixz="2.93681e-08" iyy="0.00025387" iyz="-4.26218e-07" izz="0.000154833"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_hip_3">
-      <inertia ixx="3.27938e-05" ixy="2.67901e-07" ixz="-4.41463e-09" iyy="5.69957e-05" iyz="3.32338e-08" izz="3.26449e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_ankle_2">
-      <inertia ixx="0.000335682" ixy="-7.53335e-08" ixz="2.93681e-08" iyy="0.00025387" iyz="-4.26218e-07" izz="0.000154833"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_shoulder_1">
-      <inertia ixx="2.24541e-05" ixy="-1.65173e-08" ixz="-3.55626e-06" iyy="2.29582e-05" iyz="-6.13337e-08" izz="1.00116e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_shoulder_2">
-      <inertia ixx="0.000158626" ixy="-1.12091e-07" ixz="1.02574e-05" iyy="0.000206526" iyz="1.19785e-06" izz="0.000104956"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_shoulder_3">
-      <inertia ixx="0.000142333" ixy="3.60361e-07" ixz="4.94445e-05" iyy="0.000197746" iyz="2.65919e-07" izz="0.000151821"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_elbow_1">
-      <inertia ixx="8.13344e-05" ixy="7.36974e-06" ixz="-1.04536e-05" iyy="5.86947e-05" iyz="3.92498e-06" izz="9.68353e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_wrist_1">
-      <inertia ixx="5.57722e-06" ixy="-3.10462e-08" ixz="-9.44554e-07" iyy="5.29845e-06" iyz="-1.59685e-07" izz="1.87724e-06"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="r_hand">
-      <inertia ixx="0.000162381" ixy="-2.3536e-06" ixz="3.61526e-05" iyy="0.00043138" iyz="2.07146e-05" izz="0.00030575"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_shoulder_1">
-      <inertia ixx="2.24676e-05" ixy="-1.77064e-08" ixz="3.55951e-06" iyy="2.29712e-05" iyz="6.57597e-08" izz="1.00139e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_shoulder_2">
-      <inertia ixx="0.000159621" ixy="1.19701e-07" ixz="-1.29687e-05" iyy="0.000222574" iyz="1.20112e-06" izz="0.000120056"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_shoulder_3">
-      <inertia ixx="0.00014245" ixy="-2.22441e-07" ixz="-4.95846e-05" iyy="0.000197807" iyz="1.07538e-07" izz="0.000151706"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_elbow_1">
-      <inertia ixx="8.13352e-05" ixy="-7.37159e-06" ixz="1.04505e-05" iyy="5.86823e-05" iyz="3.9319e-06" izz="9.68469e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_wrist_1">
-      <inertia ixx="5.56774e-06" ixy="3.08362e-08" ixz="9.42153e-07" iyy="5.28838e-06" iyz="-1.58924e-07" izz="1.87579e-06"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="l_hand">
-      <inertia ixx="0.000162491" ixy="2.96453e-06" ixz="-3.68906e-05" iyy="0.000430657" iyz="2.06885e-05" izz="0.000304729"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="neck_1">
-      <inertia ixx="4.0647e-05" ixy="8.09156e-08" ixz="-3.74998e-08" iyy="3.881e-05" iyz="9.39717e-06" izz="4.7535e-05"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>

<link name="neck_2">
-      <inertia ixx="0.000117943" ixy="-5.25591e-07" ixz="4.68307e-07" iyy="4.04013e-05" iyz="9.81476e-06" izz="0.000105186"/>
+      <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>
```

nunoguedelha, 2017-10-30:
@diegoferigo @traversaro Edited the checklist in the initial comment of this issue.

S-Dafarra, 2017-11-02:
> Both the wrist yaw joints go in hardware fault. Apparently the position control is unstable and it brings the joints toward the limits. I tried to reduce the gains, but it did not work

Apparently my tests weren't that good. Changing the PIDs does the job! Setting 0 for the derivative and the integral part seems to be enough.

gabrielenava, 2017-12-07:
## iCub models meeting

This afternoon @nunoguedelha, @diegoferigo, @traversaro  and I had a meeting about how to use iCub models. These are the outcomes of our meeting:

#### Definition of the new workflow for models

- All robots .urdf **raw models** are autogenerated inside `icub-models-generator` repository. Then, they are copy-pasted inside `icub-models` repository. A .urdf model for simulation is added to the list of models. **Status: already implemented**

- All composite models (e.g. icub on a chair, icub seesaw...) are inside `icub-gazebo-wholebody` repository. **Status: almost finished**

- `icub-gazebo` repository still contain an old version of icub .sdf, but it will be deprecated soon. **Status: tell people we're going to deprecate icub-gazebo**

- With the introduction of WBToolbox 3.0, `yarp-wholeBodyInterface` is now deprecated.

#### TODO list:

- simplify collision meshes for the feet

- add estimated joint friction to the model

- gain tuning of the new model for balancing with WBT3.0

- use directly raw models into Gazebo

- overwrite initial joint config. in the composite models

nunoguedelha, 2017-12-08:
Edited meeting minutes https://github.com/robotology-playground/icub-model-generator/issues/52#issuecomment-350012659: converted "TODO list" to a check list.

fiorisi, 2018-05-08:
@nunoguedelha can you please update the issue?

traversaro, 2020-03-20:
The only issue still open related to this issue is https://github.com/robotology/icub-model-generator/issues/56 . However, that issue does not prevent to use the models in Gazebo, and so I think we can close this issue.

closed 2020-03-20 by traversaro