robotology/icub-models-generator #57: Handle colors in sdf files

diegoferigo, 2017-07-24:
Fixing https://github.com/robotology-playground/icub-model-generator/issues/56 should be enough to handle the `urdf` side, adding something similar to the model:

**simmechanics:**
```xml
<MaterialProp>
    <color>0.666667,0.666667,0.666667</color>
    <transparency>0</transparency>
</MaterialProp>
```

**urdf:**
```xml
<material name="r_forearm_color">
    <color rgba="0.666667 0.666667 0.666667 1.0"/>
</material>
```

However, this doesn't seem to be enough for Gazebo, which uses [OGRE colors](http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros#Materials:Usingpropercolorsandtextures). In my opinion using the default colors is not general enough, and it would require setting a kind of map between what is exported from the CAD (or specified in the simmechanics file) to the OGRE color keywords. One of the possible option to overcome this would be adding into the generator toolchain the possibility to generate a custom [color file](https://bitbucket.org/osrf/gazebo/src/default/media/materials/scripts/gazebo.material?fileviewer=file-view-default#gazebo.material-129) and export it where gazebo can find it.

Eventually, these lines are required in the **urdf** for the usage in gazebo:

```xml
<gazebo reference="r_forearm">
    <material>r_forearm_color</material>
</gazebo>
```

cc @traversaro @nunoguedelha 

### Resources:

* [Materials: Using proper colors and textures](http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros#Materials:Usingpropercolorsandtextures)
* [Building a Visual Robot Model with URDF from Scratch](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#Material_Girl)

traversaro, 2017-07-24:
Perhaps it is not document, but apparently the SDF format has the capability of specifying the colors of the material directly in the in the `ambient`/`diffuse` / `specular` / `emissive` of the `material` tag: http://sdformat.org/spec?ver=1.6&elem=material . 

This capabilities seem to be  alternative to the `script` tag, at least from the description of the script tag: "Name of material from an installed script file. This will override the color element if the script exists."  Assuming that by "color" element they are assuming the `ambient`/`diffuse` / `specular` / `emissive`.

However I think it is worth to open an issue in https://bitbucket.org/osrf/sdformat/issues about the confusing documentation of the `script` tag, i.e. referencing a "color" element that does not exists in SDF.

Other related issues: 
* https://bitbucket.org/osrf/sdformat/issues/86/incorrect-urdf-to-sdf-parsing-materials
* https://bitbucket.org/osrf/sdformat/issues/132/parser-does-not-handle-urdf-material
* https://bitbucket.org/osrf/sdformat/issues/155/support-data-uri-for-meshes-materials
* https://bitbucket.org/osrf/sdformat/issues/128/override-arbitrary-parts-of-included

diegoferigo, 2017-07-24:
That way would avoid the installation of an additional script file, simpler is better, true. Although it is not trivial to my eyes how to convert those properties (examples in the [Heidelberg model](https://github.com/robotology/icub-gazebo/blob/master/iCubHeidelberg01_model/iCubHeidelberg01_SIM_MODEL.xml#L598)), in the Gazebo values ([example here](https://bitbucket.org/osrf/gazebo/src/default/media/materials/scripts/gazebo.material?fileviewer=file-view-default#gazebo.material-129)).

traversaro, 2017-07-24:
The documentation of the data contained in the MaterialProp tag is given in http://it.mathworks.com/help/physmod/smlink/ref/pmitvismatprop.html . 
I am not an expert, but I think those format can be converted in the format taken by Gazebo by multiplying the "intensity" (from 0.0 to 1.0) of each component of the light for the material color for the first three elements of the Gazebo "color", and the transparency (fourth element of each Gazebo "color") can be copied directly from the transparency MaterialProp element.

lucaTagliapietra, 2017-07-25:
From the one-shot experience I had the only time I tried it, I can confirm your thoughts @traversaro

diegoferigo, 2017-07-28:
https://github.com/robotology/simmechanics-to-urdf/pull/28 includes the support to gazebo colors elements as XML blobs.

As a reference, this is the **simmechanics** code:

```xml
<MaterialProp>
    <color>0.666667,0.666667,0.666667</color>
    <ambient>0.4</ambient>
    <diffuse>0.85</diffuse>
    <specular>0.9</specular>
    <shininess>0.3</shininess>
    <transparency>0</transparency>
    <emission>0.35</emission>
</MaterialProp>
```

That is translate to this **urdf** code:

```xml
<link name="r_forearm">
...
  <visual>
    ...
    <material name="r_forearm_color">
      <color rgba="0.666667 0.666667 0.666667 1.0"/>
    </material>
  </visual>
  ...
</link>
...
<gazebo reference="r_forearm">
  <visual>
    <material>
      <ambient>0.2666668 0.2666668 0.2666668 1.0</ambient>
      <diffuse>0.56666695 0.56666695 0.56666695 1.0</diffuse>
      <specular>0.6000003 0.6000003 0.6000003 1.0</specular>
    </material>
  </visual>
</gazebo>
```

The last snipper of code is then included in Gazebo directly into the `<visual>` element of the sdf (note that the method specified in the top post is deprecated. The preferred way is this one).

closed 2017-07-28 by diegoferigo

traversaro, 2017-07-28:
> note that the method specified in the top post is deprecated. The preferred way is this one 

Where did you find this information?

diegoferigo, 2017-07-28:
> Where did you find this information? 

It was my very personal understanding of the comments in the [parser_urdf.cc](https://bitbucket.org/osrf/sdformat/src/654ab4c930e1abe28208df47d4a49d8cf7515e08/src/parser_urdf.cc?at=default&fileviewer=file-view-default#parser_urdf.cc-1914). I didn't spend much time on it though.