robotology/icub-models-generator #62: The generated simmechanics doesn't contain MaterialProp elements

diegoferigo, 2017-07-28:
*Moved from https://github.com/robotology/simmechanics-to-urdf/issues/27.*

For some reason, the steps that generate the XML simmechanics file don't export anymore the MaterialProp elements. Without this, the generated urdf has no color.

@traversaro @fiorisi Who can help me on this? I have 0 knowledge about how Simscape Multiboby Link works.

traversaro, 2018-02-12:
As Simscape Multibody Link is a closed source software, I think it is quite difficult to fix this. Consider also that we are currently using a legacy format in the converter, that is not considered the main output of the conversion tool anymore.

diegoferigo, 2018-02-12:
I never opened the software. Is it somehow configurable? Probably we changed a setting without noticing. If I recall well, the file from where the `icub-gazebo` model is generated contains those fields.

traversaro, 2018-02-12:
The amount of configuration is extremely limited: 

[image]

If it ever worked (and I am not sure about this, to be honest) it was  probably broken by either an update in Creo or Matlab.

traversaro, 2018-02-12:
See http://wiki.icub.org/wiki/Creo_Mechanism_to_URDF for more info.

diegoferigo, 2018-03-27:
You can find a recap about the steps for coloring the model at https://github.com/robotology-playground/icub-model-generator/issues/57.

Actually the random colors are read properly by RViz, obtaining a very colorful and ugly iCub. Probably the best way to go now is disabling the colors if they are not exported from Link as @traversaro suggested. If it is easy enough we can manually color the meshes (if they are split wisely) @fiorisi. 

cc @lucatagliapietra @claudia-lat

fiorisi, 2018-03-28:
As far as I know from the CAD model is not possible to extract the colors using the SimMechanics Link.
Theoretically we could generate the meshes separately like it was done for R1. With iCub is a bit more complex though because we don't have the native cover surfaces.