robotology/icub-models-generator #65: URDF models deployed automatically

fiorisi, 2017-08-23:
Dear all,
 recently, together with @traversaro, we started to have a look at the automatic generation of the URDF models hosted in the repository [icub-models][1] using our [CI system][4] [Travis][2].
The goal is to have the URDF models deployed automatically every time a change is committed to the [icub-model-generator][3] repository.

For this purpose, it is possible to configure Travis such that at the end of the build process it commits the generated files.  
Hereafter is described one possible approach to automate the process:

1. Create a GitHub `BOT` account.
2. Invite the `BOT` account to the [icub-models][1] and [icub-model-generator][3] repositories
3. Generate a [personal access token][5] for the `BOT` account
4. Add the personal access token as an [Environment Variable][6] in the [Travis settings][7] of the icub-model-generator repository
5. Add the following code to the [`before_script`][10] section of the [Travis configuration file][8]
```
# clone the output folder and add the environment variable
  - git clone https://github.com/fiorisi/icub-models
  - cd icub-models
  - export ICUB_MODELS_SOURCE_DIR=`pwd`
  - cd ../
```
6. modify the [`cmake`][9] command to include the `ICUB_MODELS_SOURCE_DIR`
```
  - cmake -D ICUB_MODELS_SOURCE_DIR=$ICUB_MODELS_SOURCE_DIR  ..
```
7. At the end of the [`script`][11] section add a line to call the target to copy the new files
```
  - sudo cmake --build . --target copy-models-to-icub-models
```
8. Add a `after_success` section with the following lines
```
after_success:
  # Start in the home directory
  - cd ../..
  # Commit the new models back to the icub-models repository
  - cd icub-models
  - git remote rm origin
  - git remote add origin https://BOT_USER_NAME:${BOT_TOCKEN}@github.com/robotology-playground/icub-models.git
  - git status
  - git add .
  - git commit --message "Travis build ${TRAVIS_BUILD_NUMBER}"
  - git push --set-upstream --quiet origin master
```
Where:
* `BOT_USER_NAME` is the account of the GitHub `BOT`.
* `BOT_TOCKEN` is the name of the Travis Environment Variable created before

A preliminary test has been done on a [fork][12] of the Robotology Playground repositories. As you can see in this [commit][13] the system is working.

As suggested by @traversaro, it would be nice to have a copy of the message associated to the commit that triggered the build. Furthermore, might be also useful to have the version of all the libraries used during the conversion process, and probably there should be a check that allows the commit only if the master branch was modified.  
If you think that this automation can be useful, we can have a discussion to implement it properly.

I cc the main contributors of the repositories but feel free to add who you think might be interested in this topic.

@traversaro, @francesco-romano, @diegoferigo, @nunoguedelha, @pattacini, @DanielePucci 

[1]:https://github.com/robotology-playground/icub-models
[2]:https://travis-ci.org
[3]:https://github.com/robotology-playground/icub-model-generator
[4]:https://en.wikipedia.org/wiki/Continuous_integration
[5]:https://help.github.com/articles/creating-a-personal-access-token-for-the-command-line/
[6]:https://docs.travis-ci.com/user/environment-variables/#Defining-Variables-in-Repository-Settings
[7]:https://travis-ci.org/robotology-playground/icub-model-generator
[8]:https://github.com/robotology-playground/icub-model-generator/blob/master/.travis.yml
[9]:https://github.com/robotology-playground/icub-model-generator/blob/master/.travis.yml#L46
[10]:https://github.com/robotology-playground/icub-model-generator/blob/master/.travis.yml#L6
[11]:https://github.com/robotology-playground/icub-model-generator/blob/master/.travis.yml#L48
[12]:https://github.com/fiorisi/icub-model-generator
[13]:https://github.com/fiorisi/icub-models/commit/f01abf9e2aeee04ba9e7703024a9647dd9b16823

diegoferigo, 2017-08-24:
@fiorisi I [recently discovered](https://github.com/loc2/jekyll-template/blob/c40d69103eb0d00891e5b7f02e25fb55892f9c91/README.md#this-repo) the `git subtree` command for pushing a subfolder to a separate branch (an orphan branch of the same repository). Probably using this way and adding a second remote to `icub-model-generator` may simplify the deployment of the autogenerated folder, and the commit uploaded will have [the same name](https://github.com/loc2/jekyll-template/commits/gh-pages) of the last one from the initial repository.

I'm not expert on this, but it could be worth investigating.

fiorisi, 2017-08-25:
@diegoferigo that would be great!
Do you think that we can try to implement it on the [forked repository][1]?

[1]:https://github.com/fiorisi/icub-model-generator

diegoferigo, 2017-08-25:
I just realized that with git subtree you can commit to a second repository files that have been already committed somewhere. This is not the case since the folder containing the model is **not** present in `icub-model-generator`, and is grabbed from the `build` folder by a custom CMake target.

fiorisi, 2017-08-25:
So can we proceed with the solution proposed in the first comment?

diegoferigo, 2017-08-25:
@fiorisi I would go with something similar:

```
after_success:
  # Commit the new models back to the icub-models repository
  - cd $ICUB_MODELS_SOURCE_DIR
  - git remote add origin-botlogin https://BOT_USER_NAME:${BOT_TOKEN}@github.com/robotology-playground/icub-models.git
  - git status
  - git add --all
  - git commit -a -m "Travis build ${TRAVIS_BUILD_NUMBER}" -m "Commit: $TRAVIS_COMMIT"
  - git push origin-botlogin master
```
@traversaro any better idea?

fiorisi, 2017-08-25:
Thanks @diegoferigo. Some other open questions:
* Who has the rights to create the `BOT` and invite it to the repositories?
* How can we name the `BOT`?

diegoferigo, 2017-08-25:
> Who has the rights to create the BOT and invite it to the repositories?

ping @DanielePucci 

> How can we name the BOT?

It would be nice finding a funny name ^^ But if we are sad persons something like `loc2-bot` might work for me.

diegoferigo, 2017-08-25:
Important thing, I forgot. We must restrict the `after_success` step to only commits targeting `master` (and also not PRs). For doing this we can exploit the [`TRAVIS_BRANCH`](https://docs.travis-ci.com/user/environment-variables/#Default-Environment-Variables) variable

pattacini, 2017-08-25:
@diegoferigo 

We have already a [bot](https://github.com/LOC2Bot)

Unfortunately, its name is not so fancy 😉

fiorisi, 2017-08-25:
Can we use the `LOC2 Bot` also for `robotology-playground`?

pattacini, 2017-08-25:
@fiorisi of course!

fiorisi, 2017-09-06:
implemented with 70e7137909419bb51027aeef512e5e3a1f0f4f79.

closed 2017-09-06 by fiorisi