robotology/icub-models-generator #77: Change feet collision meshes to a simpler primitive shape

nunoguedelha, 2017-11-09:
The collision meshes in the foot should be substituted with a simpler primitive shapes (box or four spheres). This would improve the simulation performance and reduce the computations for estimating the contact forces.

fiorisi, 2017-11-12:
The iCub foot has a quite complex shape. The SDF format provides 3 possible viable solutions to approximate the collision geometry.

### [Sphere][1]
In theory, we can have 4 or more little spheres on each corner of the foot. The major drawback, in this case, is that collisions that don't involve the spheres are not detected (i.e. the foot volume can intersect with a step, if the step edge is in contact only with the center of the sole).
[image]

 
### [Box][2]
In this case, we can either draw a box smaller than the foot or a bit bigger.
[image]
[image]

### [Polyline][3]
Starting from SDF version 1.5, it is possible to use also polylines. In this case we can better approximate the foot shape. If we go for this geometry, we need to update the [supported geometries][4] in the [simmechanics-to-urdf][5] tool.
[image]
[image]

[1]:http://sdformat.org/spec?ver=1.6&elem=geometry#geometry_sphere
[2]:http://sdformat.org/spec?ver=1.6&elem=geometry#geometry_box
[3]:http://sdformat.org/spec?ver=1.6&elem=geometry#geometry_polyline
[4]:https://github.com/robotology/simmechanics-to-urdf/blob/master/simmechanics_to_urdf/firstgen.py#L37
[5]:https://github.com/robotology/simmechanics-to-urdf

nunoguedelha, 2017-12-10:
We decided to go for the use of polylines for defining the collision geometry. As a reminder: the library package GTS has to be installed. It has been checked in macos that the library can be installed with Gazebo by upgrading to latest version 8.1.1.7 through Homebrew.
The **simmechanics-to-urdf** dependency is implemented by ...

nunoguedelha, 2017-12-11:
### Background
#### Collision geometry and Polyline definition in the SDF model
A link can contain multiple collision elements (refer to `collision` definition in [this Gazebo tutorial](http://gazebosim.org/tutorials?tut=build_model)). For instance:
```
<link name='l_foot'>
  ...
  <collision name='l_foot_collision'>
      ...
  </collision>
  <collision name='l_foot_collision_2'>
      ...
  </collision>
  ...
</link>
```
In each collision element, only the first defined geometry element will be considered, the remaining ones ignored. 

#### Collision geometry and Polyline definition in the URDF model
As per the URDF specification [here](http://wiki.ros.org/urdf/XML/link), Polyline shapes are not supported in URDF format.
[Multiple collision bodies](http://wiki.ros.org/urdf/XML/link) are not supported in URDF either. Custom collision elements, for instance meant for collision checking by a controller, can be added wrapped in a custom tag—<collision_XXXXX>—which will be ignored by the ROS URDF parser.

#### Supported URDF to SDF conversions
Polyline shapes are ignored by the Gazebo parser if embedded in a URDF file. For this reason, the Polyline parameters must be embedded in a `<gazebo>` tag.

nunoguedelha, 2017-12-11:
### Requirements

#### Desired resulting collision geometry in SDF format
- The overall desired foot collision geometry would be defined in the URDF or explicitely in a SDF block wrapped in a tag `<gazebo>`
- the geometry is composed by the union of:
    - the upper part of the sole mesh
    - the extruded polyline shaping the contact with the ground

We give [here](https://github.com/robotology-playground/icub-model-generator/wiki/Example-models-(URDF,-SDF)-of-a-standalone-foot-link-with-simplified-collision-geometry#collision-shape-based-on-multiple-collision-elements) an example of a model composed by a standalone foot link with simplified collision geometry. The extruded polyline geometry element between the foot mesh and the ground has 4 points and an exagerated height for making it more visible.
<attached image>

#### YAML config parameters
The Polyline shapes are required only for iCub Gazebo models so they could be inserted through Cmake while generating the YAML config file, as for the chest assigned mass `@CHEST_ASSIGNED_MASS@` or the links assigned inertias `@GAZEBO_ASSIGNED_INERTIAS@` variables. Considering the current partial support of polylines, it would be preferable to use the polylines only on iCub Gazebo models. The input parameters to define are as follows:
```
point: [<x1>,<y1>]
point: [<x2>,<y2>]
...
height: <h>
```
#### Derived requirements
Considering the constraints on defining Polylines in URDF and on the `<gazebo>` tag behavior, the retained solution would be to define two new links "l_sole" and "r_sole", wrapped in a `<gazebo>` tag, including only a `collision` and `inertial` blocks (no visual block) and rigidly fixed respectively to the left and right foot links.

fiorisi, 2017-12-11:
Thanks @nunoguedelha for the analysis. Do you think that we need the collision in the URDF? Otherwise, for the time being, I would keep is simple: use the polyline as the collision geometry of the sole for all the models (also the one that we don't use in Gazebo).

nunoguedelha, 2017-12-11:
### Solution 2 Requirements

#### Foot link change in the URDF 
Replace the collision mesh by the box defined through the YAML config parameters.

#### YAML config parameters
As for the previous solution, the collision box is required only for iCub Gazebo models so the parameters could be inserted through Cmake when generating the YAML file from the YAML template. The input parameters would be generated in the YAML file as:
```
collisionShapes:
  - linkName: l_foot
    type: box
    xyz: [<x>,<y>,<z>]
    originXYZ: [<x>,<y>,<z>]
    originRPY: [<r>,<p>,<y>]
  - linkName: r_foot
    ...
```

nunoguedelha, 2017-12-11:
> Thanks @nunoguedelha for the analysis. Do you think that we need the collision in the URDF? Otherwise, for the time being, I would keep is simple: use the polyline as the collision geometry of the sole for all the models (also the one that we don't use in Gazebo).

Well, if we go for the Polyline choice (solution 1), it requires defining two new links "l_sole" and "r_sole" that we fixe to the feet. restraining it to Gazebo models through Cmake is no big deal and avoids adding complexity (additional links) to the other models.
Consider that sooner or later, Polylines should be supported also in URDF, or at least by the Gazebo URDF parser (unless they don't use any layer over the `urdfdom` parser), and only then we can replace the mesh or add to it the polylines within the foot link definition.

The biggest simplification would be to use just a box for now, on all models or just the Gazebo model.

nunoguedelha, 2017-12-11:
If we go for solution 1, we can roceed as follows:
- replace the feet mesh by a box on all models
- add a second geometry in the foot collision element labeled `<collision_controller_check>` that would be ignored by Gazebo, and could be parsed any of our controllers if required.

traversaro, 2017-12-12:
> Consider that sooner or later, Polylines should be supported also in URDF, or at least by the Gazebo URDF parser (unless they don't use any layer over the urdfdom parser) 

I am afraid Gazebo will never support Polylines in URDF files.

nunoguedelha, 2017-12-17:
After last discussion with @fiorisi , we decided to go for solution 2, until a better support of polylines is available:
- replace the feet mesh by a box on all models
The `simmechanics-to-urdf` parser already supports replacing meshes with simple geometric shapes, as it was tested on R1.

fiorisi, 2018-01-05:
Possible sole approximation (the foot frame is relative to iCub 2.5).
[image]
[image]

fiorisi, 2018-01-09:
Solved with https://github.com/robotology-playground/icub-model-generator/commit/bc1ed7765d51be14a323b02cbe1936644e029d22.

closed 2018-01-09 by fiorisi

traversaro, 2019-01-13:
In the future, to do this automatically (and more importantly, systematically) it could make sense to investigate the use of approximate convex decomposition techniques such as https://github.com/kmammou/v-hacd/ .

traversaro, 2019-01-13:
cc @aerydna @claudiofantacci @xEnVrE