robotology/icub-models-generator #89: Error while loading iCubGazeboV2_5 inGazebo

gabrielenava, 2018-05-08:
I downloaded [icub-models]() and I tried to load the model in Gazebo. The robot collapsed on the floor. Here is the error that I see in the terminal:

```
[DEBUG]Requested couplings: COUPLING (matrixJ2M (1.0 0.0 0.0 0.0 -1.625 1.625 0.0 0.0 0.0 0.0 1.625 0.0 0.0 0.0 0.0 1.0) (1.0 0.0 0.0 0.0 1.0 0.0 0.0 -1.0 1.0)) 
[DEBUG]Size:  2 
[ERROR]Unknown coupling type 
yarpdev: ***ERROR*** driver <gazebo_controlboard> was found but could not open
[ERROR]GazeboYarpControlBoard : controlBoard < right_arm_no_hand > did not open. 
```

That seems related to the wrong usage of `couplings` keyword in the controlboard plugins.

traversaro, 2018-05-08:
Keeping this open to find a proper solution per @nunoguedelha .

fiorisi, 2018-05-08:
As discussed with @nunoguedelha, we close the issue.
@nunoguedelha will open a new PR to reintegrate the "coupling matrices".

closed 2018-05-08 by fiorisi