robotology/icub-models-generator #90: Errors while inserting iCubGazeboV2_5 in Gazebo

gabrielenava, 2018-05-17:
My laptop configuration is:

- `yarp`, `icub-main`, `GazeboYarpPlugins`, in `devel` branch.

- Gazebo version v9.0.

As soon as I inserted the `iCubGazeboV2_5` model in Gazebo, on the Gazebo terminal an endless list of errors appeared and Gazebo crashed. Here is the error:

``` cmake
[ERROR]CBW( /icubSim/left_arm ):  getMotorEncoderAccelerations  on device left_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoders  on device right_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_arm ):  getMotorEncoders  on device left_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_arm ):  getMotorEncoders  on device right_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_arm ):  getMotorEncoderSpeeds  on device right_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_arm ):  getMotorEncoderAccelerations  on device right_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoders  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderSpeeds  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderAccelerations  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_arm ):  getMotorEncoderSpeeds  on device left_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoderSpeeds  on device right_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_arm ):  getMotorEncoderAccelerations  on device left_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoderAccelerations  on device right_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoders  on device right_leg returns  true or interface not availables [ERROR]CBW( /icubSim/left_leg ):  getMotorEncoders  on device left_leg returns  true or interface not availables 

[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderSpeeds  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoderSpeeds  on device right_leg returns  true or interface not availables [ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderAccelerations  on device left_leg returns  true or interface not availables 
[ERROR
][ERRORCBW( /icubSim/right_arm ):  getMotorEncoders  on device right_arm_no_hand returns  true or interface not availables 
]CBW( /icubSim/right_leg ):  getMotorEncoderAccelerations  on device right_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_arm ):  getMotorEncoderSpeeds  on device right_arm_no_hand returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_arm ):  getMotorEncoderAccelerations  on device right_arm_no_hand returns  true or interface not availables 
[[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoders  on device left_leg returns  true or interface not availables 
ERROR][ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderSpeeds  on device left_leg returns  true or interface not availables 
CBW( /icubSim/right_leg ):  getMotorEncoders  on device right_leg returns  true or interface not availables 
[ERROR[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoderSpeeds  on device right_leg returns  true or interface not availables 
]CBW( /icubSim/left_leg ):  getMotorEncoderAccelerations  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/right_leg ):  getMotorEncoderAccelerations  on device right_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoders  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderSpeeds  on device left_leg returns  true or interface not availables 
[ERROR]CBW( /icubSim/left_leg ):  getMotorEncoderAccelerations  on device left_leg returns  true or interface not availables 
```

traversaro, 2018-05-17:
I think this is related to https://github.com/robotology/yarp/pull/1668 .

DanielePucci, 2018-05-17:
Probably @diegoferigo can take over on this?

traversaro, 2018-05-17:
cc @valegagge

valegagge, 2018-05-18:
This issue is not a bug, but only an annoying prints...and more over not so clear! :(

The issue is due that now controlBoardWrapper prints an error if the device doen't implement the interface  to which the called function belongs. (in our case IMotorEncoder::getEncoders)
Infact the motion control device of Gazebo doesn't implement IMotorEcnoder interface and ControlBoardWrapper calls IMotorEncoder->getEncoders in the run method in order to publish motor ecoder values on stateExt port.

However In order to mantain the old behaviour I'll remove this prints.

valegagge, 2018-05-18:
see https://github.com/robotology/yarp/commit/0c0de4a9331b9b843ac4b3d3746c074dd0427249

traversaro, 2018-08-23:
Fixed by https://github.com/robotology/yarp/commit/0c0de4a9331b9b843ac4b3d3746c074dd0427249 , closing.

closed 2018-08-23 by traversaro