robotology/icub-models-generator #91: STRAIN2 bno055 orientation

traversaro, 2018-05-23:
In the model generated using simmechanics, we do not have the frames corresponding to the bno055 sensors mounted on the new STRAIN2 electronics. As adding them to the CREO cad will take time (if it is worth at all) it would be already helpful if we could extract the transform between the sensor frame  of the bno055 documented in Page 24 of https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST_BNO055_DS000_14.pdf and the FT sensor frame defined in http://wiki.icub.org/wiki/FT_sensor . 

If the STRAIN2 electronics is already model in the iCub's CAD, I think this should be trivial.

traversaro, 2018-05-23:
I extracted manually for now the transform. 
It is: 
~~~
{}^{ft_sensor_frame} H_{bno055_sensor_frame} 
= 

~~~
see the attached image for reference. 
[image]

traversaro, 2018-05-23:
cc @hu-yue

traversaro, 2018-05-23:
cc @fiorisi

hu-yue, 2018-07-16:
cc @nunoguedelha [here](https://github.com/robotology/icub-model-generator/issues/91#issuecomment-391323610) is the transformation between the IMU and the FT.

LOC2Bot, 2018-07-31:
cc @nunoguedelha and @fiorisi as discussed F2F, we should regenerate the URDF with the sensor frames.

fiorisi, 2018-08-02:
As mentioned F2F the frames are not in the FT sensors used in the CAD model of iCub. More in details, in the CAD model are used the old FT sensor (1st generation) that don't have the new strain2 board.
In order to generate the model we have to:
- [ ] talk with mech. department to understand if we can replace the FT sensors in the iCub and Koroibot CAD models
- [ ] replace the FT sensors in the iCub and Koroibot CAD models
- [ ] check that the mass properties of the FT are consistent (i.e. the geometries in the electronic board model are all solid)
- [ ] add the measurement frame in the FT sensor model (is not defined)
- [ ] redefine the [Simplified Representation](http://support.ptc.com/help/creo/creo_optm/usascii/index.html#page/optm/options_modeler/asm_two_sub/About_Creating_Simplified_Representations.html) of the links that include the FT sensors (and check all the others)
- [ ] add the frames to be exported in the CAD model
- [ ] regenerate the [ShrinkWraps](http://support.ptc.com/help/creo/creo_optm/usascii/index.html#page/optm/options_modeler/asm_three_sub/About_Shrinkwrap_Models.html) of the links that include the FT
- [ ] replace the new links in the Creo Mechanism model
- [ ] export the Creo Mechanism model to SimMechanics
- [ ] update the `.yaml` file
- [ ] regenerate the `URDF` model

fiorisi, 2018-08-13:
The sensor is located on the `TOP` face of the strain2 board (yellow pins in the figure).
[image]

To correctly define the orientation of the frame we need to locate the `PIN 1` in the chip (in the manual is highlighted with the black dot).
[image]

From the electronic schematics we see that is the lower right pin.
[image]

This is confirmed by the small white dot on the chip.
[image]

The frame has been placed in the center of the chip (on the upper surface).
[image]

fiorisi, 2018-08-13:
If the placement of the sensor is correct, the transform is different from the [one obtained by @traversaro](https://github.com/robotology/icub-model-generator/issues/91#issuecomment-391323610).

[image]

nunoguedelha, 2018-08-13:
Hi @fiorisi I've crosschecked and find the same orientation as you did, as long as the board has the default configuration P1, among the possible values below (`AXIS_REMAP_CONFIG`, `AXIS_REMAP_SIGN` registers):
[image]

traversaro, 2018-08-18:
Thanks @fiorisi for the check. I am not sure were I did the error, probably it was when placing the frame in the CAD as the document linked in  https://github.com/robotology/icub-model-generator/issues/91#issuecomment-391323851 contains the dot of the U10 board in the right position. 

@hu-yue this was probably affecting the results of the RPY analysis of the IMU output.

fiorisi, 2018-08-23:
The FT are connected to the [`EMS`](http://www.icub.org/images/brochures/iCub_controllers_flyer_web.pdf) boards. The frames names in the CAD model will be named as follows (similar to the convention used for the gyro and accelerometers):
SCSYS_`L/R` _ `BODY-PART`_ EMS_FT-IMU_`ELECTRICAL-NAME`

The `ELECTRICAL-NAME` is documented [here](https://github.com/robotology/icub-model-generator/files/2313474/iCub.3_3.0.2_Motor.Board.Placement.pdf).

fiorisi, 2018-08-23:
Here the names of the frames in the CAD model:
SCSYS_L_UPPER_LEG_FT-IMU_3B12
SCSYS_R_UPPER_LEG_FT-IMU_3B11
SCSYS_L_FOOT_FT-IMU_3B13
SCSYS_R_FOOT_FT-IMU_3B14
SCSYS_L_UPPER_ARM_FT-IMU_1B6
SCSYS_R_UPPER_ARM_FT-IMU_2B6

nunoguedelha, 2018-08-23:
> The FT are connected to the EMS boards.

As a side note, it's a pitty the number of the EMS is not in the FT board electronic name, as it is the case for the other sensors. For instance : 
- board MTB 10B2 is the MTB board #2 connected to the EMS eb10, and thus named: `SCSYS_L_UPPER_LEG_MTB_ACC_10B2`,
- Gyro directly placed on the EMS eb4, named: `SCSYS_CHEST_EMS_GYRO_EB4`.

fiorisi, 2018-08-24:
@nunoguedelha, I actually wanted to add it but the name of the features in Creo is limited to 31 characters.

fiorisi, 2018-08-24:
In the iCub Mechanism CAD model the FT sensor frames of the arms are defined in the link the contains the FT sensor itself. 

[image]

As specified in the documentation of the [simmechanics-to-urdf](https://github.com/robotology/simmechanics-to-urdf):

> Note that for now the FT sensors sensor frame is required to be coincident with child link frame, due to URDF limitations.

Currently there is a `frameReferenceLink` parameter to remap the frame to the correct link, but while redefining the Mechanism model, we can also move the FT frames in the CAD model and update the simmechanics-to-urdf [configuration file](https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in).

traversaro, 2018-08-26:
> > Note that for now the FT sensors sensor frame is required to be coincident with child link frame, due to URDF limitations.

The actual limitation is in the SDF, in particular it is tracked by this issue: https://bitbucket.org/osrf/sdformat/issues/130/position-part-of-force_torque-sensor-pose .

fiorisi, 2018-08-27:
The mass of the following links is different. I am trying to understand what happened. Maybe I will release a first version with the different masses. Please remember to double check that the simulations behavior is not different and to check that visually the inertia box is similar to the old ones (inside gazebo).

| LINK NAME | Mass difference c.a. [kg] | COMMENTS | 
| :------------: | :-------------------------: | :------ |
| SIM_SEA_2-5_L_ELBOW | 0.06 | Inertia tensor is also very different |
| SIM_SEA_2-5_R_ELBOW | 0.06 | Inertia tensor is also very different |

fiorisi, 2018-10-05:
PR opened: https://github.com/robotology/icub-model-generator/pull/111

cc @FabioBergonti

traversaro, 2020-01-24:
This was fixed by https://github.com/robotology/icub-model-generator/pull/122, thanks to everyone that contributed.

closed 2020-01-24 by traversaro