robotology/icub-models-generator #92: asHomogeneousTransform() returns incorrect Homogeneous transformation matrix from the model

vibhoraggarwal, 2018-06-12:
**Description:**
- When calculating homogeneous transformation matrix with [model.urdf](https://github.com/robotology-playground/icub-models/blob/master/iCub/robots/iCubGenova02/model.urdf) file, for the tranformation from the sensor frame(`l_leg_ft_sensor`) to base frame(`root_link`) using [getRelativeTransform()](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1KinDynComputations.html#ae9f6e6896de67e035b3fc079a8938065) followed by [asHomogeneousTransform()](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1Transform.html#a8db2358dc66b742a7dcc19e5e43f4e39) we don't get the matrix that we expect
- The issue deals with the checking in iDyntree code and make improvements as necessary

vibhoraggarwal, 2018-06-12:
Following is the matrix it currently returns: 
\begin{equation}\^{rootlink}H_{llegftsensor}=[\begin{matrix}
-0.0000&0.9880&-0.1547&-0.0042 \\\\
1.0000&0.0000&0.0000&-0.0701 \\\\
0.0000&-0.1547&-0.9880&-0.1935 \\\\
0&0&0&1.0000 \\\\
\end{matrix}\] \tag{1}\end{equation}
The images below describe the root frame, and the way the frame is generally described for FT Sensor. 
[image][image]
@traversaro

traversaro, 2018-06-12:
Thanks @vibhoraggarwal . 
As  first step, I will add a new test in https://github.com/robotology-playground/icub-model-generator/blob/master/tests/icub-model-test.cpp#L285 to make sure that, at least in the models generated by this repo, the transform between the F/T sensors and the root_link is the one that we expect.

fiorisi, 2018-06-17:
Thanks @vibhoraggarwal.

> we don't get the matrix that we expect

Which is the matrix that you would expect instead?

vibhoraggarwal, 2018-06-17:
@fiorisi As the FT sensor on the leg is oriented with its wire connection on the rear, so the homogenous transformation matrix could ideally be represented in two ways:(here `a,b,c` represents the translational parameters between the `root_link` and the `l_leg_ft_sensor`)
1. When facing upwards                                                                   
[image] $\^{root\\\_link}H_{l\\_leg\\_ft\\_sensor}=  \begin{bmatrix}-1& 0  &   0  &   a\\\\
     0  &   -1 &    0   &  b\\\\
    0 &    0   &  1  &  c\\\\
     0   &  0   &  0   &  1\end{bmatrix}$

2. When facing downwards
[image] $\^{root\\\_link}H_{l\\_leg\\_ft\\_sensor}=  \begin{bmatrix}-1& 0  &   0  &   a\\\\
     0  &  1 &    0   &  b\\\\
    0 &    0   & -1  &  c\\\\
     0   &  0   &  0   &  1\end{bmatrix}$
But, the [matrix currently obtained](https://github.com/robotology-playground/icub-model-generator/issues/92#issuecomment-396659155) represents, as if, the orientation follows the order: 

- Translation in x,y and z axis$\rightarrow$Rotationabout_z(-90$^{\circ}$)$\rightarrow$Rotationabout_y(180$^{\circ}$)$\rightarrow$Rotationabout_x(8.8851$^{\circ}$). It means that the FT sensor is facing downwards, with its wire connection on the left side of the robot; which, as seen from the robot is incorrect.

P.S: We are here interested in rotation which is what we know definitely; thus, I talk in terms of rotation and show translation as variables.

fiorisi, 2018-06-18:
@vibhoraggarwal you are correct. The error is in the CAD model, probably is my mistake.
Good catch!

[image]

traversaro, 2018-06-18:
The issue was confirmed by the test I added in https://github.com/robotology/icub-model-generator/commit/44380b8aef7dc87450307a56a54264105e82d2c8 . The result on the iCubGenova04 model contained in the repo is the following:

~~~
    Start 9: iCubGenova04ConsistencyCheck
9: Test command: /home/straversaro/src/icub-model-generator/build2/tests/icub-model-test "--model" "/home/straversaro/src/icub-model-generator/build2/iCub/robots/iCubGenova04/model.urdf"
9: Test timeout computed to be: 1500
9: icub-model-test : base_link test performed correctly 
9: icub-model-test : sole are parallel test performed correctly 
9: icub-model-test : transform between root_link and l_leg_ft_sensor is not the expected one, test failed.
9: icub-model-test : Expected transform : -1 0 0
9: 0 1 0
9: 0 0 -1
9: 
9: icub-model-test : Actual transform : -4.68977e-12 1 2.06823e-13
9: 1 4.68977e-12 9.69952e-25
9: 0 2.06823e-13 -1
9: 
1/1 Test #9: iCubGenova04ConsistencyCheck .....***Failed    0.04 sec
~~~

The same test running on the iCubGenova04 model contained in `yarp-wholebodyinterface` has this output:
~~~
straversaro@iiticublap103:~/src/robotology-superbuild/robotology/yarpWholeBodyInterface/app/robots/iCubGenova04$ /home/straversaro/src/icub-model-generator/build2/tests/icub-model-test "--model" model.urdf
icub-model-test : base_link test performed correctly 
icub-model-test : sole are parallel test performed correctly 
Check for model model.urdf concluded correctly!
~~~

I guess the issue was introduced sometime after the last synch of the model in yarp-wholebodyinterface, that happened (I guess) around January 2017 : https://github.com/robotology/yarp-wholebodyinterface/commit/fe3d454375fed650f333ca2ea13979a758c3acb4#diff-502483deb5879b243007f9c45652d5bb .

fiorisi, 2018-06-20:
@vibhoraggarwal the CAD model has been updated and the new file commited. As soon as the new URDF model is deployed to the [models](https://github.com/robotology/icub-models) repository can you check if the issue is solved?
In case, feel free to close this issue.

traversaro, 2018-06-20:
cc @fjandrad note that there was an error in the FT sensor frame in models contained in icub-models since 1.5 year, while the one contained in yarp-wholebodyinterface where working correctly. Which one did you use for your tests?

vibhoraggarwal, 2018-06-20:
@fiorisi The returned homogeneous transformation matrix is still erroneous.

vibhoraggarwal, 2018-06-25:
@fiorisi I checked the new `homogeneous transformation matrix` using new [model.urdf](https://github.com/robotology/icub-models/blob/master/iCub/robots/iCubGenova02/model.urdf) file. 
The output transformation matrix is:
\begin{bmatrix}-1.0000  &  0.0000 &  -0.0000 &   0.0073 \\\\
     0.0000   &  1.0000   & -0.0000  &  -0.0701\\\\
    0  &  -0.0000   & -1.0000   & -0.1942\\\\
   0    &      0      &    0    & 1.0000\end{bmatrix}$
which is consistent with the expected matrix value in the [comment above](https://github.com/robotology/icub-model-generator/issues/92#issuecomment-397888040). 
As, the correction is successful; thus, closing the issue.

closed 2018-06-25 by vibhoraggarwal

fjandrad, 2018-06-27:
For the tests before September last year I'm not sure. I remember just copying what was being used in the robot when we switched to the green. At some point I gave a try to the new model generated, but I did not used it as the main model. But since september/october I have been using the model with the skin frames that @fiorisi helped create.
Checking for the relative transform between root link and left leg sensor I get:
[image]
Which matches the result @vibhoraggarwal  was expecting.

traversaro, 2018-06-27:
@fjandrad great.