robotology/icub-models-generator #98: Generate 3d taxels positions for iCub's sole taxels

traversaro, 2018-07-11:
This would be useful for @hu-yue's experiments. 

We had some preliminary data obtained in 2016 by @prashanthr05 available in https://github.com/robotology-playground/icub-taxel-positions , but it would be good if we could add the triangle centers frames in the CAD models to generate the taxel positions using the pipeline of this repo, described in https://redmine.robotology.eu/easy_knowledge_stories/57 . The sole are actually quite convenient for this process because the patch is completely flat and the **triangles are already modeled at the CAD level**. 

* [ ] Generate a pptx file containing the numbering of the triangles in the sole path (see  https://redmine.robotology.eu/s/D2WH329 on how to generate these from the ini generators available in https://github.com/robotology/icub-main/tree/master/app/skinGui/iniGenerators )
* [ ] Add the frames in Creo model 

cc @DanielePucci @fiorisi

fiorisi, 2018-07-11:
If someone can take care of the `pptx` files with the triangle numbers, I can add the frames in the CAD model. It might take some time then to update also the [simmechanics-to-urdf files](https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in). Probably the model could be ready by the beginning of the next week.

cc @fjandrad

traversaro, 2018-07-11:
If someone wants to produce the  pptx, please check https://redmine.robotology.eu/s/D2WH329 for instructions on how to produce the pptx from the foot ini generators: 
* https://github.com/robotology/icub-main/blob/cd2028daac5afcddb220f3b6a7bfe6caa4abdf7b/app/skinGui/iniGenerators/left_foot_ini_generator.xlsx 
* https://github.com/robotology/icub-main/blob/cd2028daac5afcddb220f3b6a7bfe6caa4abdf7b/app/skinGui/iniGenerators/right_foot_ini_generator.xlsx 
For the photos to annotate please ask to  Marco Maggiali or Davide Gandini.

hu-yue, 2018-07-13:
We asked Marco Maggiali and he gave us the numbering of the feet soles skin patches. 
I updated the files in `icub-main`, as these are files that were originally stored in the `skinGui` conf files to generate the ini files, but were a bit difficult to interpret. 
Marco added the numbers in the file.
I opened a pull request to merge these files into the `devel` branch of `icub-main`: https://github.com/robotology/icub-main/pull/540.

fiorisi, 2018-07-16:
The CAD, [simmechanics and generation files](https://github.com/robotology/icub-model-generator/tree/soles-skin-frames/simmechanics/data/icub2_5) have been updated (thanks @fjandrad for the help!).
Waiting for the [green flag](https://travis-ci.org/robotology/icub-model-generator/builds/404509575?utm_source=github_status&utm_medium=notification) from Travis. 
TODO:
- [ ] compute the taxels position with the [Python script](https://github.com/robotology/icub-model-generator/tree/master/simmechanics/scripts)

hu-yue, 2018-07-17:
@fiorisi and @fjandrad thanks! I will try to generate the taxel position with the script.

hu-yue, 2018-07-17:
Sorry but I tried and I have some questions.
So if I got it correctly, the procedure should:
- Generate the URDF with:
```
simmechanics_to_urdf ICUB_2-5_plus_BB_SIM_MODEL.xml --output xml --yaml ICUB_2-5_BB_simmechanics_options.yaml.in --csv-joints ICUB_2-5_BB_joint_parameters.csv.in --outputfile iCub_2_5.urdf
```
- Generate the taxels position file with:
```
./generate_taxels_3d_positions.py --urdf iCub_2_5.urdf --link l_sole --skin_frame l_sole --skinGui_conf_file left_foot.ini --indipendent_patches --skinManager_conf_file ../../generated/skinGui/conf/positions/l_foot.txt --plot
```

I had issues generating the URDF:
```
Traceback (most recent call last):
  File "/usr/local/bin/simmechanics_to_urdf", line 9, in <module>
    load_entry_point('simmechanics-to-urdf==0.2', 'console_scripts', 'simmechanics_to_urdf')()
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 1871, in main
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 160, in convert
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 289, in parseYAMLConfig
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 257, in fetch_more_tokens
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning for the next token
found character '@' that cannot start any token
  in "ICUB_2-5_BB_simmechanics_options.yaml.in", line 11, column 1
```

So I guess I am probably doing something wrong.
Then I have doubts about the second step:
- The ini file for the skin is the one inside `icub-main`, right? (Those [here])(https://github.com/robotology/icub-main/tree/cd2028daac5afcddb220f3b6a7bfe6caa4abdf7b/app/skinGui/conf/skinGui)
- The `skin_frame` should be what exactly? Is this also referred to the URDF model?

Are there any guidelines about this? I could not find much, could you give me some hints? @fiorisi @fjandrad @traversaro?
Thanks!!

traversaro, 2018-07-17:
The `.yaml` file of simmechanics should be the one generated by CMake in https://github.com/robotology/icub-model-generator/blob/master/simmechanics/CMakeLists.txt#L196 , the one you are using is the template, and so python is complaining for the presence of `@`.

hu-yue, 2018-07-17:
Ok I see, so I had to build icub-model-generator to get the new urdf files.
Thanks @traversaro.
However, when I tried to build it but I get this error:
```
In file included from /home/yuehu/Software/iCub_SW/icub-model-generator/tests/icub-model-test.cpp:5:0:
/home/yuehu/Software/iCub_SW/robotology-superbuild/build/install/include/iDynTree/Model/Indeces.h:15:4: warning: #warning <iDynTree/Model/Indeces.h> is deprecated. Please use <iDynTree/Model/Indices.h>. To disable this warning use -Wno-deprecated. [-Wcpp]
   #warning <iDynTree/Model/Indeces.h> is deprecated. Please use <iDynTree/Model/Indices.h>. To disable this warning use -Wno-deprecated.
    ^
/home/yuehu/Software/iCub_SW/icub-model-generator/dh/generator/urdf_utils.cpp:16:49: fatal error: iDynTree/ModelIO/impl/urdf_import.hpp: No such file or directory
compilation terminated.
dh/generator/CMakeFiles/urdf_from_dh_utils.dir/build.make:62: recipe for target 'dh/generator/CMakeFiles/urdf_from_dh_utils.dir/urdf_utils.cpp.o' failed
make[2]: *** [dh/generator/CMakeFiles/urdf_from_dh_utils.dir/urdf_utils.cpp.o] Error 1
CMakeFiles/Makefile2:1110: recipe for target 'dh/generator/CMakeFiles/urdf_from_dh_utils.dir/all' failed
make[1]: *** [dh/generator/CMakeFiles/urdf_from_dh_utils.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

It did generate some urdf files, but I didn't see the soles skin anywhere.
So I tried to use `simmechanics_to_urdf` to generate it but I get this error:
```
Traceback (most recent call last):
  File "/usr/local/bin/simmechanics_to_urdf", line 9, in <module>
    load_entry_point('simmechanics-to-urdf==0.2', 'console_scripts', 'simmechanics_to_urdf')()
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 1871, in main
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 167, in convert
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 850, in buildTree
Exception: ('The frame: ', 'SCSYS_L_ANKLE_ROLL_FRAME', ' doesn t lay on the joint axis.')
```

traversaro, 2018-07-17:
Disable the `ICUB_MODEL_GENERATE_DH` option, thanks.

hu-yue, 2018-07-17:
I eventually succedeed in generating the taxels positions :D
As @traversaro suggested, I had to change all the `frameReferenceLink` tht were set to `r_sole` and `l_sole`, to `r_foot` and `l_foot`, as the soles are just frames and not links (?).

With these commands:
- right foot
```
./generate_taxels_3d_positions.py --urdf ../../build/iCub/robots/iCubGenova04/model.urdf --link r_sole --skin_frame r_sole --skinGui_conf_file ../../../icub-main/app/skinGui/conf/skinGui/right_foot.ini  --indipendent_patches --skinManager_conf_file ../../generated/skinGui/conf/positions/r_foot.txt --plot
```
- left foot
```
./generate_taxels_3d_positions.py --urdf ../../build/iCub/robots/iCubGenova04/model.urdf --link l_sole --skin_frame l_sole --skinGui_conf_file ../../../icub-main/app/skinGui/conf/skinGui/left_foot.ini  --indipendent_patches --skinManager_conf_file ../../generated/skinGui/conf/positions/l_foot.txt --plot
```
[image]
[image]

I will prepare a pull request with the modifications and the generated txt files with taxels positions.

traversaro, 2018-07-17:
As you may notice, there is an approximation that is currently used that is particularly bad for the taxels in the foot: currently the taxels that are outside the convex hull of the are approximated with the triangle center (see https://github.com/robotology/icub-model-generator/blob/master/simmechanics/scripts/generate_taxels_3d_positions.py#L215 for the code related to this).

hu-yue, 2018-07-17:
yes I was wondering why there are no taxels outside, this is actually pretty bad, considering that in the feet the taxels that are the most active are those on the boundaries.

fiorisi, 2018-07-17:
The [pull request](https://github.com/robotology/icub-model-generator/pull/100) has been merged. The new models should be automatically commited to the [icub-models](https://github.com/robotology/icub-models) repository soon.
@hu-yue feel free to close this issue.

hu-yue, 2018-07-18:
@fiorisi I created a new null request to update the yaml.in file, otherwise with the current version the generated URDF would not take into account the new frames.
See https://github.com/robotology/icub-model-generator/issues/98#issuecomment-405618444.

traversaro, 2018-07-18:
> yes I was wondering why there are no taxels outside, this is actually pretty bad, considering that in the feet the taxels that are the most active are those on the boundaries. 

For 2D patches probably it just make sense to get the 3D centers of the triangles from the URDF model and generate the rest of the taxel position by just adding the offset w.r.t. of each triangle, considering the orientation of the patch plane and the angle of the triangle, adapting to the new script the idea implemented by @prashanthr05 in https://github.com/robotology-playground/icub-taxel-positions/blob/master/cad_extraction/r_foot/foot_taxel.py#L135 .

fiorisi, 2018-07-18:
@traversaro consider that the orientation of the frame was defined as follows: 

> each coordinate frame shall be oriented with the z axis normal to the skin surface (pointing outward) and x oriented along the main dimension of the skin-covered link (direction is arbitrary, but shall be consistent)

traversaro, 2018-07-18:
> @traversaro consider that the orientation of the frame was defined as follows:

Sorry, with orientation I meant the 2D angle provided by the iCubSkinGui configuration file, not the orientation of the frame, i.e. the `orient` parameter, see https://github.com/robotology/icub-model-generator/blob/master/simmechanics/scripts/generate_taxels_3d_positions.py#L330 .

fiorisi, 2018-09-07:
Can we close this issue?

fjandrad, 2018-09-08:
Did anyone tried this?

> For 2D patches probably it just make sense to get the 3D centers of the triangles from the URDF model and generate the rest of the taxel position by just adding the offset w.r.t. of each triangle, [...]

traversaro, 2018-09-08:
Not that I am aware of.

hu-yue, 2018-09-08:
I didn't try that. For me we can close this issue for the time being.

closed 2018-09-10 by fiorisi

traversaro, 2018-09-11:
I just realized that a super-easy way to achieve extrapolation is to use [KD-Tree](https://en.wikipedia.org/wiki/K-d_tree) (or similar NN-friendly data structures) for finding the closest neighbors instead of [Delanauy triangulation](https://en.wikipedia.org/wiki/Delaunay_triangulation). Once you have the 3 (or 4 or more) 3D  nearest neighbors, you can choose any interpolant surface  (in a linear case, a plane) to approximate the surface and  even extrapolate the data. One of the advantages of this is that the libraries for performing implementing KD-Tree or similar data structure are usually available in most programming languages, and are more common than libraries implementing delanauy triangulation (finding a suitable library was one of the problems for @joankangro ).

See https://ieeexplore.ieee.org/document/7877140/ . 

cc @fjandrad @matejhof

traversaro, 2018-09-12:
Note that, for internal poitsn, nearest neighbors is equivalent to Delanauy triangulation only if the point you are interpolating from are sufficiently regular (otherwise you could have a lot of  close points on "one side", and the point you want to interpolete could be outside the convex hull of the 3/4 nearest neighbors.

traversaro, 2018-09-12:
> Can we close this issue?

Sorry for the late reply: I think the issue  should not be closed, as the taxel position were generated, but they are not available in any repository, neither this one nor in https://github.com/robotology/icub-main/tree/master/app/skinGui/conf/positions . However, I will open a new issue for this for clarity.

matejhof, 2018-09-12:
Hi everyone,
for our Nao Prague with iCub skin, I had a student working on getting the taxel pos in 3D over the summer. We did not get far with interpolation - the reason could be that we have quite small patches with 8-16 triangles and quite big curvatures.
What we ended up doing was overlaying the 2D patch over the 3D triangle pos from CAD in Matlab and then rotating the normals triangle by triangle. 
[image]
[image]