robotology/icub-models-generator #114: FT IMU frames not generated in the simmechanics model for 2-5_plus models

prashanthr05, 2019-08-08:
The`ICUB_2-5_plus_BB_SIM_MODEL.xml` do not have the FT IMU frames generated. This causes errors while trying to add the FT IMU frames in the `ICUB_2-5_BB_simmechanics_options.yaml.in` file

While compiling, I get this error:
``` bash
[  3%] Generating iCubGazeboV2_5_plus.urdf
Traceback (most recent call last):
  File "/usr/local/bin/simmechanics_to_urdf", line 11, in <module>
    load_entry_point('simmechanics-to-urdf==0.2', 'console_scripts', 'simmechanics_to_urdf')()
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 1871, in main
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 177, in convert
  File "build/bdist.linux-x86_64/egg/simmechanics_to_urdf/firstgen.py", line 249, in addSensors
KeyError: ('r_hip_2', 'SCSYS_R_UPPER_LEG_FT-IMU_3B11')
simmechanics/CMakeFiles/generate-models-simmechanics.dir/build.make:92: recipe for target 'simmechanics/iCubGazeboV2_5_plus.urdf' failed
make[2]: *** [simmechanics/iCubGazeboV2_5_plus.urdf] Error 1
CMakeFiles/Makefile2:1202: recipe for target 'simmechanics/CMakeFiles/generate-models-simmechanics.dir/all' failed
make[1]: *** [simmechanics/CMakeFiles/generate-models-simmechanics.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

## REASON 
The IMU frames have not been generated in the `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml` and in the options file `icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in`, we are asking the `firstgen.py` parser to add the IMU sensors by accessing the relevant frame names in the generated sim mechanics file.

## WORKAROUND
- To test if the changes in the PR are passing, I commented out the macro in `generate_icub_simmechanics` that uses the `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml`. With this change, the compilation succeeds.

- Maybe a proper fix could be to maintain two yaml option files, one for `ICUB_2-5_plus_BB_SIM_MODEL.xml` and another for `ICUB_2-5_BB_SIM_MODEL.xml`.

Any suggestions @nunoguedelha @fiorisi ?

_Originally posted by @prashanthr05 in https://github.com/robotology/icub-model-generator/pull/113#issuecomment-519510998_

nunoguedelha, 2019-08-08:
I wouldn't go for that solution (2 `yaml` option files) to avoid duplicating information while keeping things simple enough.

### While the `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml` model is not properly updated...
We can add the parameters related to the FT-IMU via the `CMakeLists.txt` as it was done through the cmake variables `@XSENS_IMU_FRAME@`, `@XSENS_IMU_SENSOR@` and using the condition tag `GIVTWO_ICUB_PLUS`:
https://github.com/robotology/icub-model-generator/blob/bada6e53f0c874d80812a0b3cd146145d1ea85c9/simmechanics/CMakeLists.txt#L131-L142

### Regarding the `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml` model update...
The FT-IMU frames are useful when the FT sensors are equipped with the **strain2** boards, but when the FT sensors still have the old strain boards, having those IMU frames in the model does not pose a problem. Anyway, this is independent from the additional features integrated in the cub2.5plus model (inclined soles, XSENS IMU sensor).
So `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml` should be updated with those FT-IMU frames, unless we decide to drop that model all-together, which is another discussion.

nunoguedelha, 2019-08-08:
CC @traversaro @fjandrad

nunoguedelha, 2019-08-08:
As as side note, it is the same situation with the XSENS IMU sensor I would say. The frame `SCSYS_ROOT_LINK_XSENS_IMU` should be present in both models `icub2_5/ICUB_2-5_plus_BB_SIM_MODEL.xml` and `icub2_5/ICUB_2-5_BB_SIM_MODEL.xml` so the later should be updated.

prashanthr05, 2019-08-08:
I think I accidentally deleted `@XSENS_IMU_FRAME@` from the yaml file. I will add it.

traversaro, 2019-08-09:
I agree with @nunoguedelha suggestion to use custom params to avoid mantaining to `.yml` files. Eventually if the 2.5+ model gets regenerate, we could probably avoid to have to deal with this additional parameter.

nunoguedelha, 2019-08-09:
> I agree with @nunoguedelha suggestion to use custom params to avoid mantaining to `.yml` files. Eventually if the 2.5+ model gets regenerate, we could probably avoid to have to deal with this additional parameter.

custom params added to CMakeLists.txt in https://github.com/robotology/icub-model-generator/pull/113.

traversaro, 2020-05-26:
Was this fixed by https://github.com/robotology/icub-model-generator/issues/114 @prashanthr05 @nunoguedelha ?

prashanthr05, 2020-05-26:
This was fixed by adding the custom `@FT_IMU_SENSORS@` param to  https://github.com/robotology/icub-model-generator/blob/9f6dbe7ab98c4a4762317e2c9aacb298ffe39c53/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in#L923

which was set empty for ICUB_2-5_plus models

https://github.com/robotology/icub-model-generator/blob/9f6dbe7ab98c4a4762317e2c9aacb298ffe39c53/simmechanics/CMakeLists.txt#L142

and set relevantly for ICUB_2-5 models
https://github.com/robotology/icub-model-generator/blob/9f6dbe7ab98c4a4762317e2c9aacb298ffe39c53/simmechanics/CMakeLists.txt#L177

traversaro, 2020-05-26:
Ok, so we can close this issue?

prashanthr05, 2020-05-26:
I renamed the issue to be more meaningful. We can close it.

closed 2020-05-26 by traversaro

nunoguedelha, 2020-05-26:
Just to complete the info:
- this was originally fixed in PR https://github.com/robotology/icub-model-generator/pull/113 .
- Which was closed and PRed again (for Travis issues) in https://github.com/robotology/icub-model-generator/pull/122 .