robotology/icub-models-generator #124: Add logic to commit models generated in the devel branch to the devel branch of icub-models

traversaro, 2019-10-02:
The PR https://github.com/robotology/icub-model-generator/pull/123 introduces a change that is only compatible with gazebo-yarp-plugins 3.3 (currently in the devel branch). For this reason, we need to introduce the `devel` branch also in this repo and in icub-models, and add a bit of generation on the Travis script that automatically pushes the models to icub-models, to push the generate models to the `devel` branch of icub-models when a new commit is pushed to the icub-model-generator's `devel` branch.

traversaro, 2020-09-29:
Actually, now that I think of this more, it seems to be more aligned with our workflow that we keep a single branch on this repo (i.e. `master`) and then push any change from it to the `devel` branch of icub-models. We can then manually merge from `devel` to `master` of icub-models whenever we do a new release, so approximately every three months.

What do you think of this @Nicogene ?

Nicogene, 2020-09-29:
Sounds good to me!

traversaro, 2020-10-17:
This change was implemented in https://github.com/robotology/icub-model-generator/pull/141, and it already saved us to deploy a bad regression for iCub2 models in stable branches as described in https://github.com/robotology/icub-models/issues/50 .

closed 2020-10-17 by traversaro