robotology/icub-models-generator #130: Decide where to store the hand_mk3 model

fiorisi, 2020-04-29:
The iCubTech team Dev would like to discuss where to store the files used to generate the new hand model and where to store the URDF model in [icub-models](https://github.com/robotology/icub-models).

Concerning the source files one possibility is to create a folder named `hand_mk3` in this directory:
https://github.com/robotology/icub-model-generator/tree/master/simmechanics/data

Concerning the URDF model if we store the file in the [iCub](https://github.com/robotology/icub-models/tree/master/iCub) folder we might have a mess inside of the `conf` and `meshes` subfolders.
One solution is to create another folder `A` in the [repository root](https://github.com/robotology/icub-models). Inside `A` we create a dedicated folder `hand_mk3` that contains the URDF model with the `conf` and `meshes` subfolders.

If this solution is viable, I am wondering how we can name the folder `A`. If we want to use it to store also the URDF of other robot parts or gadgets maybe we can call it `components`.

Finally, do you think that the purpose of the repository `icub-models` is still preserved?

cc @traversaro @gabrielenava @DanielePucci @pattacini @ale-git @lornat75

traversaro, 2020-04-29:
I like your proposal @fiorisi , and `components` seems to be a good name to me.

fiorisi, 2020-06-25:
The PR has been opened. Closing this discussion for the time being.

closed 2020-06-25 by fiorisi

fiorisi, 2021-02-03:
cc @vvasco @Nicogene @vtikha