robotology/icub-models-generator #179: Reenable the icub-model-test on iCub3

GiulioRomualdi, 2020-11-23:
As discussed with @traversaro we should reenable the test on icub3 in order to avoid regressions such as https://github.com/robotology/icub-models/issues/70. 
This [`model`](https://github.com/GiulioRomualdi/icub-models/blob/icub3/wbd_rebased/iCub/robots/iCubGazeboV3/model.urdf) as desiderata for the joint direction. 

cc @traversaro and @Nicogene

traversaro, 2020-11-23:
In particular, it is important because of how the Creo/Simmechanics generation works, the direction of the exported axis and hence the list of joints to keep in the [`reverseRotationAxis`](https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/ICUB_3_all_options.yaml#L417) list will change, and without a test that checks if the axis direction is correct, we never know if the model is correctly regenerate. 

An example is the PR https://github.com/robotology/icub-models-generator/pull/172, that contained a new generated model, that cause the regression described in https://github.com/robotology/icub-models/issues/70 .

GiulioRomualdi, 2020-12-07:
Hi @traversaro and @Nicogene 
Is there any plan for this?  I'm really afraid that robotology/icub-models#70 will happen again in the future

traversaro, 2020-12-11:
Hi @GiulioRomualdi, I am not aware if this is already planned to work, probably @Nicogene can provide more info on this.

closed 2020-12-22 by Nicogene