robotology/icub-models-generator #197: Increase the damping of iCubGazebov3 and the PID gains

S-Dafarra, 2021-03-24:
The damping is currently set ot 0.06 for all the joints. This value is very small, also considering the real robot. After some tests, ``2.0`` seems a more reasonable value.

Regarding the PID, I used the following values  P 40.0  I 0.35  D 0.35 on all the joints, and seemed to be working fine, at least for 
the walking.

I can also prepare a PR if you point me to the correct files to modify.

cc @Nicogene @traversaro

Nicogene, 2021-03-25:
Hi @S-Dafarra !

The damping can be specified in the csv (e.g. https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/ICUB_3_joint_all_parameters.csv)
And the PID can be changed in all the control board ini files (see https://github.com/robotology/icub-models-generator/blob/20a9bbcfa6e962c44be22a30c531724a0227fb7e/simmechanics/data/icub3/conf/gazebo_icub_torso.ini#L40-L42 for example)

A PR is more than welcome, just remember to change all the csv files here https://github.com/robotology/icub-models-generator/tree/master/simmechanics/data/icub3, for now we are using the one with all the joints in one, but in the future we would follow the modular approach

traversaro, 2021-03-25:
> A PR is more than welcome, just remember to change all the csv files here https://github.com/robotology/icub-models-generator/tree/master/simmechanics/data/icub3, for now we are using the one with all the joints in one, but in the future we would follow the modular approach

Can't we just delete the redundant one, and once we want to go for the modular approach we take the all joint one and decompose it in the separate one? Trying to maintain two different ["non"-single sources of true](https://en.wikipedia.org/wiki/Single_source_of_truth) it tricky.

Nicogene, 2021-03-25:
> Can't we just delete the redundant one, and once we want to go for the modular approach we take the all joint one and decompose it in the separate one? Trying to maintain two different "non"-single sources of true it tricky.

I am afraid that it is not yet possible because the csv it taken by the `simmechanics-to-urdf` script with the simmechanics XML and for now we have not all the modular simmechanics XMLs and we do not know how to assemble the resulting urdfs.

Or it is possible to pass to the script more than one csv??

traversaro, 2021-03-25:
> I am afraid that it is not yet possible because the csv it taken by the `simmechanics-to-urdf` script with the simmechanics XML and for now we have not all the modular simmechanics XMLs and we do [...]

Sorry, I meant to delete the modular one, as those are the one that are the moment are not used.

Nicogene, 2021-03-25:
> Sorry, I meant to delete the modular one, as those are the one that are the moment are not used. In any case, this is not a big priority.

Yeah we can restore them later if necessary

S-Dafarra, 2021-03-25:
So the final verdict? I can remove them if you want, less files to edit :grin:

traversaro, 2021-03-25:
> 
> 
> So the final verdict? I can remove them if you want, less files to edit 😁

Let's remove them!

traversaro, 2021-03-25:
Because it seems that `ICUB3_L_LEG_SIM_MODEL.xml`, `ICUB3_R_LEG_SIM_MODEL.xml` and `ICUB3_UPPERBODY_SIM_MODEL.xml` are already lagging behind w.r.t. to `ICUB3_ALL_SIM_MODEL.xml`, so better to remove them before creating more confusion.

S-Dafarra, 2021-03-25:
I have noticed those files are reference here: https://github.com/robotology/icub-models-generator/blob/20a9bbcfa6e962c44be22a30c531724a0227fb7e/simmechanics/data/icub3/convertModel.sh#L1-L12

Should I also comment out those lines?

PR #199 (2021-03-26): Increase iCubGazeboV3 damping and PIDs
Fixes #197 

cc @traversaro @Nicogene
closed 2021-03-26 by Nicogene