robotology/icub-models-generator #200: converModel.sh script for iCub3 generates a faulty urdf model

AlexAntn, 2021-04-21:
When converting an `xml` to `urdf` file using the [simmechanics/data/icub3/convertModel.sh](https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/convertModel.sh) script, with a fresh clone of the repo, the arms of the model are poorly generated (they start in weird positions, and in hardware fault, check printscreen below).

[image]

This is not the case when, instead, running:
```bash
mkdir build &&
cd build &&
cmake .. &&
make
```

This generates a correct model. 

In which situations is the [convertModel.sh](https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/convertModel.sh) script used? Additionally, we should include a reference to the `make` method on the readme or the repository.

traversaro, 2021-04-22:
> In which situations is the [convertModel.sh](https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/convertModel.sh) script used? Additionally, we should include a reference to the `make` method on the readme or the repository.

I don't think `convertModel.sh` is ever documented, the only supported way is the one that you wrote down that you can see it is used in the CI scripts: https://github.com/robotology/icub-models-generator/blob/master/.github/workflows/ci.yml#L97 . It was added in https://github.com/robotology/icub-models-generator/pull/133 , but I think it make sense to simply remove it, what do you think @Nicogene ?

Nicogene, 2021-04-22:
I agree!

closed 2021-04-22 by traversaro