robotology/icub-models-generator #216: Investigate on how to fix the colors in the urdf models

GiulioRomualdi, 2022-01-27:
Thanks to https://github.com/robotology/idyntree/pull/961 it will be possible to display the icub models with the colors defined in the `urdf` files.

I noticed that the robot colors are "randomly" chosen when the model is generated. As a consequence, the representation of the model is a bit bizarre

[image]

This is  how TALOS is loaded by the very same visualizer 

[image]

 
Do you think it's possible to fix the color of the links?

cc @traversaro

GiulioRomualdi, 2022-01-27:
I think it's related to this: https://github.com/robotology/simmechanics-to-urdf/blob/f6055ab8d627c45138ddde46532ea42a74b48caa/simmechanics_to_urdf/firstgen.py#L24-L28

traversaro, 2022-01-28:
Yes, we can: 
* Modify `simmechanics-to-urdf` (just to set a user-specify color with an option, or just remove the random color behaviour), and switch icub-model-generator to use the new version of simmechanics-to-urdf
or:
* Create a script (probably in python as it is easier) that sets all the colors in a given URDF to a user-specified color, and run that on the icub models in the icub-model-generator CI

Both options are fine for me. We also need to identify which color we want to use.

GiulioRomualdi, 2022-01-28:
> Note that simmechanics-to-urdf already have a similar option (see https://github.com/robotology/simmechanics-to-urdf/blob/f6055ab8d627c45138ddde46532ea42a74b48caa/simmechanics_to_urdf/firstgen.py#L1116), but that would require to add a line in the yaml for each link to set a different color, so it may not be super-convenient.

I actually noticed that this dictionary is filled by the `xml` file and not by the yaml file: [From here](https://github.com/robotology/simmechanics-to-urdf/blob/f6055ab8d627c45138ddde46532ea42a74b48caa/simmechanics_to_urdf/firstgen.py#L228) -> [here](https://github.com/robotology/simmechanics-to-urdf/blob/f6055ab8d627c45138ddde46532ea42a74b48caa/simmechanics_to_urdf/firstgen.py#L569) -> [finally here](https://github.com/robotology/simmechanics-to-urdf/blob/f6055ab8d627c45138ddde46532ea42a74b48caa/simmechanics_to_urdf/firstgen.py#L593-L614)

As far as I understood the `xml` file is generated by simmechanics and it cannot be modified. 

Should I replicate the same logic for the `yaml` file?

traversaro, 2022-01-28:
> I actually noticed that this dictionary is filled by the `xml` file and not by the yaml file: [From [...]

You are right, I misunderstood the code. 

> Should I replicate the same logic for the `yaml` file?

I guess not, until we have a need for it.

GiulioRomualdi, 2022-01-28:
https://github.com/robotology/icub-models-generator/pull/217 attempts to close this issue

PR #217 (2022-01-28): Fix the colors for iCubV2_5 and iCubV3
This PR closes #216 and required https://github.com/robotology/simmechanics-to-urdf/pull/51 to properly work

In detail, this PR defines the colors for each link of the robots. I've chosen the color arbitrally let me know if you prefer a different palette.

After this PR and https://github.com/robotology/idyntree/pull/961 if iCub is loaded in the iDynTree visualizer we will get:

 
[image]
[image]

# TODO
- [x] ~~Once https://github.com/robotology/simmechanics-to-urdf/pull/51  has been merged update the CI~~ Not required since the ci already get the lates [... cut]
closed 2022-01-28 by traversaro