robotology/icub-models-generator #229: The meshes of icub3 seems not compatible with rviz2

Nicogene, 2022-12-07:
When trying to visualize the icub3 meshes on `rviz2` i got these errors:

```
[ERROR] [1670425894.974567837] [rviz2]: The STL file 'package://iCub/meshes/simmechanics/sim_icub3_torso_2_prt.stl' is malformed. It starts with the word 'solid' and also contains the word 'endsolid', indicating that it's an ASCII STL file, but rviz can only load binary STL files so it will not be loaded. Please convert it to a binary STL file.
```

`ros2 pkg list | grep iCub` works fine.

cc @pattacini @traversaro

Nicogene, 2022-12-07:
Exporting as binary seems to work thanks!

[image]

traversaro, 2022-12-07:
Should we update the Creo 2 URDF guide with this indication? The guide should be at https://github.com/icub-tech-iit/cad-mechanics-public/wiki/Creo-mechanism-to-URDF (see http://wiki.icub.org/wiki/Creo_Mechanism_to_URDF), but I can't find it anymore.

Nicogene, 2022-12-12:
@traversaro the guide has been moved [here](https://github.com/icub-tech-iit/cad-libraries/wiki/Prepare-PTC-Creo-Mechanism-for-URDF)

traversaro, 2022-12-12:
Related RViz issues:
* https://github.com/ros2/rviz/issues/712
* https://github.com/ros-visualization/rviz/pull/1365

fiorisi, 2022-12-14:
> For the future we should make sure that at some point in the model generation pipeline (either before or after the files are committed in this repo) the stl files should be converted from ascii to binary.

I will investigate if the [Simscape Multibody Link](https://www.mathworks.com/help/smlink/ref/linking-and-unlinking-simmechanics-link-software-with-proengineer.html) allows to set the `.stl` generation with the `binary` option.

Nicogene, 2022-12-23:
Here is a simple script for converting the stl: https://github.com/cmpolis/convertSTL

It seems to work fine, shall we include it in our pipeline of urdf generation?

traversaro, 2022-12-23:
> Here is a simple script for converting the stl: https://github.com/cmpolis/convertSTL
> 
> It seems to work fine, shall we include it in our pipeline of urdf generation?

Ok for me.

PR #230 (2022-12-23): Add conversion of stl files from ascii to binary
It fixes #229 

For doing the conversion I used a (modified) version of https://github.com/cmpolis/convertSTL starting from [this commit](https://github.com/cmpolis/convertSTL/commit/b353f144bd14f77f1a0f3177dcc2969a0ffcf388)

[image]
closed 2022-12-23 by Nicogene