robotology/icub-models-generator #249: Remove "plus" models

traversaro, 2023-09-15:
The `iCubGazeboV2_5_plus` and `iCubGenova04_plus` refer to models based on iCubGazeboV2_5 and iCubGenova04 with a mounted KIT_007  "Ankle for climbing stairs" (see https://icub-tech-iit.github.io/documentation/upgrade_kits/ankle_for_stairs/support/#assembly-instruction). However, as far as I know no robot has mounted that for a long time, so I propose to remove them.

@Nicogene @pattacini @maggia80 @fiorisi @DanielePucci

pattacini, 2023-09-15:
Fine with me 👍🏻

Nicogene, 2023-09-15:
> The `iCubGazeboV2_5_plus` and `iCubGenova04_plus` refer to models based on iCubGazeboV2_5 and iCubGenova04 with a mounted KIT_007 "Ankle for climbing stairs" (see [icub-tech-iit.github.io/documentation/upgrade_kits/ankle_for_stairs/support/#assembly-instruction](https://icub-tech-iit.github.io/documentation/upgrade_kits/ankle_for_stairs/support/#assembly-instruction)). However, as far as I know no robot has mounted that for a long time, so I propose to remove them.
> 
> @Nicogene @pattacini @maggia80 @fiorisi @DanielePucci

Ok for me !

maggia80, 2023-09-15:
ok

traversaro, 2023-09-15:
Unfortunatly I discovered that iCubGenova02 actually has KIT_007 mounted on it. While that robot is pratically not used for walking, unless the plus models are actually problematic  I think it is easier just to leave them.

traversaro, 2023-09-15:
I did a minor cleanup in https://github.com/robotology/icub-models-generator/pull/250 .

DanielePucci, 2023-09-17:
> Unfortunatly I discovered that iCubGenova02 actually has KIT_007 mounted on it. While that robot is pratically not used for walking, unless the plus models are actually problematic I think it is easier just to leave them.

Yes @traversaro, this is related to some of the previous work we did in:
* https://github.com/ami-iit/element_icub-at-ihmc/issues/7
* https://github.com/ami-iit/element_icub-at-ihmc/issues/38

I agree that if the models are not problematic, we may leave them there. I do not know if we may find a better label than "plus" to identify the HW modification

traversaro, 2023-09-17:
> I agree that if the models are not problematic, we may leave them there. I do not know if we may find a better label than \"plus\" to identify the HW modification

Yes, the proper way is KIT_007 . plus in the name is just legacy, but if it is not a low hanging fruit I would not rename it at the moment.

traversaro, 2023-09-17:
> Yes, the proper way is KIT_007 . plus in the name is just legacy, but if it is not a low hanging fruit I would not rename it at the moment.

Done in https://github.com/robotology/icub-models-generator/pull/250 .

traversaro, 2023-09-20:
Fixed in https://github.com/robotology/icub-models-generator/pull/250 .

closed 2023-09-20 by traversaro